NimmLor/esp8266-fastled-iot-webserver #6: "user_interface.h: No such file or directory" when compiling

FlyingT, 2019-10-27:
"user_interface.h: No such file or directory" when compiling
Arduino 1.8.8 / ESP8266 2.4.2 / FastLED 3.2.6 / Links2004/arduinoWebSockets newest version 

Any help?

NimmLor, 2019-10-28:
Weird. Have you selected the correct board in the settings?

FlyingT, 2019-11-05:
> Weird. Have you selected the correct board in the settings?

I#am using [this D1 mini clone](https://www.amazon.de/gp/product/B01N9RXGHY/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1) and selected the D1 mini preset

rml7du, 2019-12-16:
Were you able to solve this issue FlyingT? I have some problem.

FlyingT, 2019-12-17:
> Were you able to solve this issue FlyingT? I have some problem.

Nope, wasnt able to solve it

closed 2020-05-07 by NimmLor