NimmLor/esp8266-fastled-iot-webserver #8: I cannot map the sinelon effect into the matrix

tihoangyeudau, 2019-11-18:
I want to convert the sinelon effect from 1d to 2d, which means the effect runs and fade its color from leaf to leaf instead of from pixel to pixel. I spent hours writing but failed. Do you help me?

this is my incompleted code: [https://youtu.be/m34n0QJ1d7k](url)

```
const bool    kMatrixSerpentineLayout = true;

#define MAX_DIMENSION ((LEAFCOUNT>PIXELS_PER_LEAF) ? LEAFCOUNT : PIXELS_PER_LEAF)

// The 16 bit version of our coordinates
static uint32_t x;
static uint32_t y;
static uint32_t z;

void sinelon1() //XYhorizontal
{
  // a colored vertical bar sweeping back and forth horizontally, with fading trails

  // fade all the LEDs, this will result in our fading trail
  fadeToBlackBy(leds, NUM_LEDS, 20);

  // get the bar's current horizontal (x) position
  uint16_t pos = beatsin16(speed, 0, PIXELS_PER_LEAF);

  // keep track of the previous position, so we can fill in any gaps when moving at high speed
  static uint8_t prevpos = 0;

  // swap them, if needed
  if (pos < prevpos) {
    uint8_t temp = pos;
    pos = prevpos - pos;
    prevpos = temp;
  }

  CRGB color = ColorFromPalette(palettes[currentPaletteIndex], gHue, 255);

  // draw a vertical line for each position between pos and prevpos
  
    for (uint8_t y = prevpos; y <= pos; y++) {
      for (uint8_t x = 0; x < LEAFCOUNT; x++) {
          leds[XY(x, y)] = color;
    }
  }

  // keep track of the previous position
  prevpos = pos;
}

uint16_t XY( uint8_t x, uint8_t y)
{
  uint16_t i;
  if( kMatrixSerpentineLayout == false) {
    i = (y * LEAFCOUNT) + x;
  }
  if( kMatrixSerpentineLayout == true) {
    if( y & 0x01) {
      // Odd rows run backwards
      uint8_t reverseX = (LEAFCOUNT - 1) - x;
      i = (y * LEAFCOUNT) + reverseX;
    } else {
      // Even rows run forwards
      i = (y * LEAFCOUNT) + x;
    }
  }
  return i;
}
```

NimmLor, 2019-11-26:
Sorry for the late response. I would have coded like this instead of using the XY helper function:

```
    for (uint8_t y = prevpos; y <= pos; y++) {
      for (uint8_t x = 0; x < LEAFCOUNT; x++) {
          for(int i = 0;i < LEDS_PER_LEAF;i++)
             leds[LEAFCOUNT * LEDS_PER_LEAF + i] = color;
    }
```

tihoangyeudau, 2019-11-26:
> Sorry for the late response. I would have coded like this instead of using the XY helper function:
>
> ```
>     for (uint8_t y = prevpos; y <= pos; y++) {
>       for (uint8_t x = 0; x < LEAFCOUNT; x++) [...]

i tried yours but it cant work. there is blank effect. this is my code:

`void sinelon()
{
  // a colored dot sweeping back and forth, with fading trails
  fadeToBlackBy(leds, NUM_LEDS, 20);
  int pos = beatsin16(speed, 0, NUM_LEDS);
  static int prevpos = 0;
  //CRGB color = CHSV( gHue, 255, 192);
  CRGB color = ColorFromPalette(palettes[currentPaletteIndex], gHue, 255);
  for (uint8_t y = prevpos; y <= pos; y++) {
      for (uint8_t x = 0; x < LEAFCOUNT; x++) {
          for(int i = 0;i < PIXELS_PER_LEAF;i++)
             leds[LEAFCOUNT * PIXELS_PER_LEAF + i] = color;
    }
  }
 prevpos = pos;
 delay(1000 / FRAMES_PER_SECOND );
}`

WarDrake, 2020-04-08:
I don't know if you got this working, but this seems to do it for me.

void sinelon_leaf()
{
  // a colored dot sweeping back and forth, with fading trails
  fadeToBlackBy(leds, NUM_LEDS, 20);
  int pos = beatsin16(speed, 0, LEAFCOUNT);
  static int prevpos = 0;
  CRGB color = ColorFromPalette(palettes[currentPaletteIndex], gHue, 255);
  if (pos < prevpos) {
    pos *= PIXELS_PER_LEAF; 
    fill_solid((leds + pos), (prevpos - pos) + PIXELS_PER_LEAF, color);
  }
  else {
    fill_solid(leds + prevpos, (pos - prevpos) + PIXELS_PER_LEAF, color);
  }
  prevpos = pos;
  delay(100);
}

I added it as an extra pattern.

closed 2020-06-16 by NimmLor