NimmLor/esp8266-fastled-iot-webserver #14: Noob problem with the D1 Mini

MrPresident2, 2020-01-23:
I don't want to waste any one's time, I'm pretty noob and I have a problem while trying to upload the files to the Mini D1 board. Is there anyone that is kind and would like to help me solve the problem?
You can add my discord: Mr.President#1929

pattyland, 2020-01-24:
Advice: "a problem" is not a good or helpful description

MrPresident2, 2020-01-24:
This is why I asked if there's someone kind enough to help me, it's too long for writing it down.

pattyland, 2020-01-25:
I find it very impolite to discuss a problem that others may have in private.

MrPresident2, 2020-02-17:
well, when I'm pressing tools- sketch data upload I'm getting this error:  SPIFFS Error: esptool not found!

SashapounNako, 2020-02-17:
I have the same mistake, I downloaded esptool with Python thinking it would fix the problem but... No. I've been trying to do this project for several months now and I still can't get to its goal because of a problem and yet I follow the instructions to the letter, it's starting to get a bit annoying >:[.

MrPresident2, 2020-02-17:
> I have the same mistake, I downloaded esptool with Python thinking it would fix the problem but... No. I've been trying to do this project for several months now and I still can't get to its goal [...]

Well, let's hope someone here would help us.

SashapounNako, 2020-02-22:
No solution to the problem?

psu1989, 2020-02-23:
From another post: I think it is because you need the newest esptool version. Download latest release from here: https://github.com/esp8266/arduino-esp8266fs-plugin/releases
I got the same error, and I just successfully uploaded the sketch with latest version (0.5.0)

closed 2020-02-27 by MrPresident2