NimmLor/esp8266-fastled-iot-webserver #20: Problem with uploading the code.

MrPresident2, 2020-02-23:
when I'm pressing tools- sketch data upload I'm getting this error: SPIFFS Error: esptool not found!

AnthonMS, 2020-02-23:
I think it is because you need the newest esptool version. Download latest release from here: https://github.com/esp8266/arduino-esp8266fs-plugin/releases

MrPresident2, 2020-02-23:
That seems to solve that problem! thanks. but when I go to the serial monitor to see the IP of the esp I get nothing.

AnthonMS, 2020-02-23:
Remember to change baud rate to 115200 the upload speed should also be set to this.

MrPresident2, 2020-02-23:
I've changed the upload rate to 115200, but I'm not sure where is the baud rate. another question, when I write my network settings should I keep it with the "   " or delete everything and just keep the SSID name without the  " " ( example: "network" or network )

AnthonMS, 2020-02-23:
If you mean in the Secrets.h, you need to keep the quotes. So it should look like "network".
After opening the serial monitor, it should open a small window where you can send commands. In the bottom of that window, there is a dropdown menu where you select baud rate, right next to "clear output" button.
But if you aer on wrong baud reate, it should just display gibberish characters. So try pressing the reset button on the d1 mini board, after opening serial monitor.

MrPresident2, 2020-02-23:
Thank you for your kind help, I appreciate it.
so I've done that but I still get gibberish characters

AnthonMS, 2020-02-23:
Hmm, weird. Are you using a Wemos D1 Mini? or another esp8266 board?

MrPresident2, 2020-02-23:
I couldn't find Wemos D1 Mini, so I got Makerfocus D1 Mini from Amazon.

AnthonMS, 2020-02-23:
Ok, never heard of that one before. But my suggestion is then, that you try the different baud rates there are, or google what you need for that specific board.

MrPresident2, 2020-02-23:
Maybe do you have a link for a board that you know is good? maybe is a problem with the board I have.

AnthonMS, 2020-02-23:
I am using this one https://www.banggood.com/D1-mini-V2_2_0-WIFI-Internet-Development-Board-Based-ESP8266-4MB-FLASH-ESP-12S-Chip-p-1143874.html?rmmds=myorder&cur_warehouse=CN
It's also a clone, but never had a problem with them.

smoochiee, 2020-02-24:
change your esp8266 core in device manager 0.5.0...it should work

MrPresident2, 2020-02-24:
> change your esp8266 core in board manager on Arduino tools tab... 0.5.0...it should work

IDK what happened but I was able to upload the code before, after that I'm back to nothing working again hahaha

MrPresident2, 2020-02-24:
> change your esp8266 core in board manager on arduino tools tab... 0.5.0...it should work

for which one of the problems?

MrPresident2, 2020-02-25:
@AnthonMS I got new boards. I upload the code but still can't see any IP. when I go to my router I don't see any new connection there ( i could have seen the IP there if there was any connection between the board and the router). Is there any way I'm doing something wrong with the code ?

closed 2020-04-17 by MrPresident2

AnthonMS, 2020-04-17:
> @AnthonMS I got new boards. I upload the code but still can't see any IP. when I go to my router I don't see any new connection there ( i could have seen the IP there if there was any connection between the board and the router). Is there any way I'm doing something wrong with the code ?

I switched to use WLED instead, because they added grouping feature. So the LEDs are grouped together. And so much nicer interface and app for Android.