NimmLor/esp8266-fastled-iot-webserver #21: Problem when I upload the code

SashapounNako, 2020-02-23:
I have this problem and I don't know how to fix it : 
`esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

SPIFFS Upload failed!`

smoochiee, 2020-02-24:
change your esp8266 core t0 5.0

SashapounNako, 2020-02-24:
Already have 0.5.0

smoochiee, 2020-02-24:
still not working?u have add spiff on your document?

SashapounNako, 2020-02-24:
Which document?

smoochiee, 2020-02-24:
document/arduino/tools/

smoochiee, 2020-02-24:
if problem not fixed maybe you have bad d1 mini. ..you error show that module is not responding

SashapounNako, 2020-02-24:
I'll look at it when I get home. For the arduino, I just got the same one I got from Thingiverse.

SashapounNako, 2020-02-24:
Okay, so I think it's just the card that sucks because I did everything right and put them all in the right folders.
[image]

NimmLor, 2020-02-24:
Hi, you might have wrong upload settings. Please verify that the upload settings look the same as in the screenshot below (except for the port).

[image]

SashapounNako, 2020-02-26:
What the... Okay... I plugged in USB 2.0 on my motherboard and successfully uploaded.

SashapounNako, 2020-02-26:
Ah, and I reinstalled the driver.

SashapounNako, 2020-02-26:
So, now the problem is that I can't find the ESP on my network and when I click on the "reset" button with the serial monitor no IP address is displayed (I soldered the leds and they are white and when I click on "reset" I feel like they blink for a millisecond).

MrPresident2, 2020-02-27:
I have the same problem, I uploaded the code, but nothing happens and I can't see the IP or the board on my wifi.  did you solve that problem?

closed 2020-06-16 by NimmLor