NimmLor/esp8266-fastled-iot-webserver #25: Sound reactive mode not sensitive enough

BotiB, 2020-03-16:
Everything looks to work fine, my only issue is that the sound reactive sensor is not sensitive enough. Have to put music very close or very very loud to get it working. I'm using the MAX4466 microphone sensor.
Is there option in the program to make it more sensitive?

tihoangyeudau, 2020-03-17:
I think reactive sound effect is draft

NimmLor, 2020-03-17:
Yes, it's not that easy to work with sound sensors on the esp8266, therfor I'm working on an software to stream the audio data directly to the microcontroller. 

I've coded a software (for windows) that allows syncing the music with all my LED projects using an Windows audio output.

[image]

In the upcoming update (eta ~2-3 weeks) all of my projects will be merged into one.

tihoangyeudau, 2020-03-17:
Very nice . But with nanoleaf, can music effects be show on group pixels per leaf?

flashez, 2020-04-28:
> Yes, it's not that easy to work with sound sensors on the esp8266, therfor I'm working on an software to stream the audio data directly to the microcontroller.
>
> I've coded a software (for windows) [...]

Hey great projet ! Any updates since ? Will this software be released someday ?
I was looking for a way to sync my nanoleafs (your project) with my Spotify flow.

NimmLor, 2020-04-29:
Hi, the improved arduino code will be released in around 5 days, the desktop application will be added as a "work-in-progress" project in another repository soon. 

Currently I'm working on the integration of my 7-segment clock, the nanoleafs and my desk lamp will follow afterwards.
The webinterface has now a darkmode that can be toggled and buttons to enter the OTA/Alexa mode.

Sorry for the late response @tihoangyeudau :

> Very nice . But with nanoleaf, can music effects be show on group pixels per leaf?

There will be custom effects designed for the nanoleafs, in addition to the regular visualization patterns.

> Can we sync effects many devices via udp to control with one webserver?

Each LED device has to be added to the desktop software and receives the audio data via udp individually.

> So, will we still have to use pc for reactive music?

The desktop software is completely optional, however it is required to sync the leds to music from an windows audio output. Unfortunatly I'm not planning on implementing the sound-sensor stuff, because it doesn't work to good and only measures an absolute "volume" level, the windows application however sends an entire fft calculation ("volume" of each frequency band).

NimmLor, 2020-05-07:
Update is here and the audio visualization via udp was implemented. Another user improved the sound sensor values.

flashez, 2020-05-07:
> Update is here and the audio visualization via udp was implemented. Another user improved the sound sensor values.

By the way, do we need to update arduino code on every D1 Mini on each project ?

NimmLor, 2020-05-07:
You should do upload the sketch and the sketch data (aka Website) to each esp8266. The desktop software should work with the old software, but the visualization patterns aren't available.

flashez, 2020-05-07:
> You should do upload the sketch and the sketch data (aka Website) to each esp8266. The desktop software should work with the old software, but the visualization patterns aren't available.

Alright, I'll update them all. Sadly I glued my nanoleafs :(

closed 2020-05-11 by NimmLor