NimmLor/esp8266-fastled-iot-webserver #26: Board isn't connecting to WiFi/can't fine IP

MrPresident2, 2020-03-29:
Hey Guys,
after uploading the code it seems like nothing really happens. I can't find the IP address and when I look at my network I can't see any new devices at all.
I hope you guys can help me with that.
Thanks!

visualdeath, 2020-03-29:
You are not reallygiving us much information to try and help you with your issue. Can you send a serial monitor log. Have you verified the WiFi settings in your secrets.h file?

MrPresident2, 2020-03-29:
Yes, my wifi settings are correct, and I'm getting nothing on my serial monitor that's the problem.
I've been trying to do this project for the last 3 months IDK what to do and I really need someones to help

visualdeath, 2020-03-29:
You are not getting any output on the serial monitor? Is the serial monitor set on 115200 baud? What kind of board are you uploading to?

MrPresident2, 2020-03-29:
It is set on that, and I tried all the other baud options. It's a mini D.
If you have discord or something we can talk so it will be easier ill pay you for your time.

visualdeath, 2020-03-29:
https://discord.gg/7nJbE5

closed 2020-04-17 by MrPresident2