NimmLor/esp8266-fastled-iot-webserver #30: Improvement : MQTT / Home Assistant / Google Home

WarDrake, 2020-04-11:
By using MQTT protocol, the service can connect to other systems
specifically smart home controllers like Home Assistant / OpenHab / etc
from there, exposing the lights to Google Home / Siri in addition to local control by Alexa

WarDrake, 2020-04-11:
Adding the MQTT Protocol to the arduino sketch allows for connection to multiple platforms, local or cloud ones.

The MQTT protocol and channels are preconfigured for automatic discovery and configuration for Home Assistant, and exposing to Google Assistant

Available endpoints are are :
Power
Brightness
RGB
Patterns.

This does not replace Alexa integration, it expands on it, both may be active at the same time.

after exposing the device to Google Assistant from Home Assistant, Google can control by voice :
Power
Brightness
RGB (Switches to Solid Color Pattern)

to control Patterns by voice add a script to Home Assistant triggering the device in this format

'1586569671485':
alias: Rainbow Effect
sequence:

data:
effect: Horizontal Rainbow
entity_id: light.nanoleafs
service: light.turn_on
The phrase will then be "Activate " to trigger the patterns using Voice commands from google home.

WarDrake, 2020-04-11:
The pubsubclient library has an Issue sending long messages from esp8266 so this will throw up an error on compile with the standard library, to fix it, we'll need to add an overloaded function to it.

src/pubsubclient.h line 145
boolean publish(const char* topic, const char* payload, unsigned int length, boolean retained);

src/pubsubclient.cpp line 368-370
boolean PubSubClient::publish(const char* topic, const char* payload, unsigned int length, boolean retained) {
return publish(topic,(const uint8_t*)payload, length, false);
}

NimmLor, 2020-05-07:
Implemented in the newest update.

closed 2020-05-07 by NimmLor