NimmLor/esp8266-fastled-iot-webserver #32: LED glitched after the 8th Leaf

MrPresident2, 2020-04-17:
Hey guys,
I'm having a weird problem with my LEDs. After the 8th leaf, the LEDs start to glitch.
Now it happens faster in high brightness setting but it will always happen. Everything starts working fine and after a few minutes, 9-12 Leaves stop working and getting stuck on one or multiple colors. 
I've tried changing the LEDs but the problem stays the same.  I do use a logic shifter, the power supply is 5V 5A. When the LEDs stop working after the 8 leaves, the output on the data from the last strip on the 8 leaf shows 3.5v (like the main power lines) instead of 0.3v as anywhere else. I've checked for wire crossing but everything is alright. 
Hope you can help me.

WarDrake, 2020-04-17:
That can only be a lack of power, it's not a voltage issue, it's an Amp Issue, so check your power source is REALLY giving you 5A instead of 2.4 if it is, then you might have some loose connection points on your pins, check which ones have a good connection and re solder the rest
did you use paste?

flashez, 2020-06-17:
Best way is to power from the other side of the strip. You'll know if it's a power issue if it only powers again only to the first 8 leafs.

bb-Ricardo, 2021-01-17:
you still experiencing this issue or have you figured out the problem.

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo