NimmLor/esp8266-fastled-iot-webserver #34: LED Glitch in Effect Mode

LordNikooon8611, 2020-04-28:
I got a nice Problem, i got the mass variant of nanoleafs 44 leafs with 6 LEDs per Leaf so there is a Total Amount of 264 WS2812B LEDs in it -> The main Problem is there are 2 Charges of Chips and when i turn the "others" (where less in it) to a other color - the LEDs glitch until the next other LED chip, then is everthing ok.

when i give them all the color blue/green/red/... except yellow the arent glitching, is there any possible variable in it where i can say the effect mode - Just change the whole color? 

or is it possible that anyone has that problem with 2 charges of chips and solved it?

The Levelshifter is in it between row 3 & 4 (the best result) 
a "new" powering modul gets also a better result of lightning

But the glitching in Effect Mode is a little Strange

NimmLor, 2020-05-07:
I've made a possible fix to your issue. A friend of mine had a similar problem with a two different WS2812 strips, the first one worked fine and the second one glitched like hell. 

I've fixed it by adding `#define FASTLED_ALLOW_INTERRUPTS 0` as the first line in the code.

LordNikooon8611, 2020-05-07:
so the "new" version with the Software support should solve the Problem in this case? btw. nice work and thx for this Software ;) btw. the Windows software doesn't let it install there is a error occuring at the end of the installation, but i will test it again

closed 2020-06-16 by NimmLor