NimmLor/esp8266-fastled-iot-webserver #38: WiFiUDP

LordNikooon8611, 2020-05-07:
WirelessVisualizer:336: error: 'WiFiUDP' does not name a type

  WiFiUDP EthernetUdp;

  ^

WirelessVisualizer.ino: In function 'void initUdp(int)':

WirelessVisualizer:2267: error: The Udp class has been renamed EthernetUdp.
As of Arduino 1.0, the Udp class in the Ethernet library has been renamed to EthernetUdp.

  Udp.begin(port);

  ^

WirelessVisualizer.ino: In function 'bool parseUdp()':

WirelessVisualizer:2276: error: The Udp class has been renamed EthernetUdp.
As of Arduino 1.0, the Udp class in the Ethernet library has been renamed to EthernetUdp.

  int packetSize = Udp.parsePacket();

                   ^

exit status 1
'WiFiUDP' does not name a type

Ich glaube da fehlt irgendwo eine definition, Versuchs grad selber herauszufinden wo ... evtl. find ichs raus, aber, dass weisst das in der Master File sich irgendwo ein Fehler eingeschlichen hat

NimmLor, 2020-05-07:
Versuch mal auf die neueste esp8266 (2.7.x) und die neueste FastLED upzudaten. Hast du in zeile 336 das "Udp" selbst in "EthernetUdp" unbenannt?

LordNikooon8611, 2020-05-07:
Nope kam direkt nach Verify/Compile ohne mein zu tun, lt. der WiFiUDP doku müsste aber "theoretisch" die WiFiUDP.h Inkludiert werden nach der ESPWiFi wenn ich des jetzt richtig rausgelesen hab,

LordNikooon8611, 2020-05-07:
also #include <WiFiUDP.h>

brachte

Sketch uses 372167 bytes (35%) of program storage space. Maximum is 1044464 bytes.
Global variables use 48456 bytes (59%) of dynamic memory, leaving 33464 bytes for local variables. Maximum is 81920 bytes.

lads jetzt noch schnell auf den nanoleaf drauf und schau obs geht

NimmLor, 2020-05-07:
Sehr komisch, bei mir und bei meinem Kollegen funktionierts ohne <WiFiUDP.h>. Ich fügs hinzu und push gleich ins repo.

LordNikooon8611, 2020-05-07:
wart noch kurz jetzt hab ich die neue Software drauf aber die leafs reagieren auf kein kommando von der Website

NimmLor, 2020-05-07:
Falscher Pin?

NimmLor, 2020-05-07:
und hast du den DEVICE_TYPE 3 definiert? und unten dann deine Leaf einstellungen getätigt?

LordNikooon8611, 2020-05-07:
normal sollte ich Type 4 definiert haben ...

#define HOSTNAME "LEDs"				// Name that appears in your network, don't use whitespaces, use "-" instead
#define DEVICE_TYPE 4				// The following types are available

#elif DEVICE_TYPE == 4
	#define LEAFCOUNT 44              // Amount of triangles
	#define PIXELS_PER_LEAF 6        // Amount of LEDs inside 1x Tringle

1. Leaf Weiss komplett und 2. Leaf 1 LED Grün ... den Fehler hab ich bei der Alten auch mal gehabt ... aber keine Ahnung mehr wie ich das behoben hab -> in der Software hat er auch nur 32 Module erkannt (angezeigt) aber weder über Webapp noch über Software reaktion

LordNikooon8611, 2020-05-07:
> Falscher Pin?

Pin könnte es noch sein

NimmLor, 2020-05-07:
> normal sollte ich Type 4 definiert haben ...

ja stimmt, hab da was durcheinander gebracht.

Probier mal auf esp8266 v2.7.0 upzudaten und auch den FastLED auf die neueste.

> 1. Leaf Weiss komplett und 2. Leaf 1 LED Grün
Das ist normal wenn er am D4 Pin hängt, das entsteht dadurch da D4 auch für WiFi status verwendet wird und da tut sich beim Starten/Uploaden was.

LordNikooon8611, 2020-05-07:
> ja stimmt, hab da was durcheinander gebracht.
>
> Probier mal auf esp8266 v2.7.0 upzudaten und auch den FastLED auf die neueste.

naja D3 war Definiert auf D4 geht der quatsch ... vielleicht liegst an dem sonst greif ich immer noch auf das update zurück,... liegt auf jedenfall an einem der Probleme ... aber wie gesagt WiFiUDP.h hat bei mir geholfen

LordNikooon8611, 2020-05-07:
BTW Läuft nun nur mehr mit der "App" Zurechtfinden ;) Danke für die Hilfe

NimmLor, 2020-05-07:
Ja, leider bin ich noch nicht dazu gekommen dafür ne Anleitung zu schreiben. Einfach rechts oben auf das "+" oder das Network Icon klicken um Geräte hinzuzufügen.

Beim Schließen des Fensters läuft die Anwendung im Systemtray rechts unten weiter.

LordNikooon8611, 2020-05-07:
> Ja, leider bin ich noch nicht dazu gekommen dafür ne Anleitung zu schreiben. Einfach rechts oben auf das "+" oder das Network Icon klicken um Geräte hinzuzufügen.
> 
> Beim Schließen des Fensters läuft die Anwendung im Systemtray rechts unten weiter.

soweit bin ich schon nur mit der musikanpassung happerts noch =) bei Soid color macht er nur weiss schaut recht witzig aus aber das glitchen ist prinzipiell bei manchen effekten immer noch da bei der letzten reihe vor dem der Levelshifter hängt weil da die komplette reihe andre chips sind aber bei musikreaktiv fällts "nicht" mehr auf

closed 2020-06-16 by NimmLor