NimmLor/esp8266-fastled-iot-webserver #39: [Suggestion] Visualizer modes

flashez, 2020-05-07:
New audio visualization is really nice.
I love the mode "41 - Nanoleaf Rainbow Band Visualizer" but the use of a threshold might be a good idea to avoid less bright triangles.
It reacts well but by default triangles are slightly on and looks not that great.
For exemple, you set a threshold to a value higher than default. When it reaches above, the triangle can light up.

Video of "less bright" triangles : [https://www.youtube.com/watch?v=GOeTbD6uXnM](url)

NimmLor, 2020-05-08:
Might be implemented into the next update

NimmLor, 2020-06-16:
Reduced threshold to 0 in [Release 4.1](https://github.com/NimmLor/esp8266-fastled-iot-webserver/releases/tag/4.1)

closed 2020-06-16 by NimmLor

flashez, 2020-06-17:
Didn't see any changes with mode "Nanoleaf Rainbow Band Visualizer"

flashez, 2020-06-17:
I tried to change value to negative and it's working flawlessly !

[img url="https://i.imgur.com/vQZ9Xt3.png"][/img]