NimmLor/esp8266-fastled-iot-webserver #44: pm open, type:0 0 error on ModeMCU board

WarDrake, 2020-05-12:
When using a NodeMCU, the system crashes and with a pm open,type:0 0
and restarts every few minutes.
(vanilla upload Serial console log, only change from master branch is data pin for NodeMCU)

[Info] Opened the serial port - COM5

connected with WDRothdam, channel 1
dhcp client start...
ip:192.168.0.16,mask:255.255.255.0,gw:192.168.0.1
Connected! Open http://192.168.0.16 in your browser

mDNS responder started 

pm open,type:0 0

(MQTT Enabled upload Serial console log, only change from master branch is data pin for NodeMCU and MQTT enable)

[Info] Opened the serial port - COM5
ip:192.168.0.16,mask:255.255.255.0,gw:192.168.0.1
Connected! Open http://192.168.0.16 in your browser

mDNS responder started 

Connecting to MQTT...
connected 

Subscribing to MQTT Topics 

Configuration Publishing Begun
Configuration Sent
Configuration Publishing Finished
pm open,type:0 0

from the moment the pm error shows up to when the system restarts, nanoleafs become unresponsive, but changes done on the website are loaded upon restart.

I don't have a weemos D1 to try it out with it.

NimmLor, 2020-05-14:
A friend of mine has noticed this issue too, I'll have to look into it what causes that issue.

WarDrake, 2020-05-18:
I haven't figured out what causes the error, but I figured out the crash and random restart issue, it seems they are unrelated

Line 20 :
#define FASTLED_ALLOW_INTERRUPTS 0

if that line is commented out, the pm message shows on the console, but it doesn't affect the system, the nanoleafs have been working flawlessly for 24 hours now.

NimmLor, 2020-05-18:
Ok that's weird, without the `FASTLED_ALLOW_INTERRUPTS 0` some older LED-Strips flicker and glitch insane. That line fixed all glitching issues for a friend of mine.

NimmLor, 2020-05-18:
I've searched a bit and found that it might be crash because the sdk doesn't get enough cpu time. The `FASTLED_ALLOW_INTERRUPTS 0` blocks the wifi handling when the LEDs are controlled by the FastLED library.

A possible fix could be to decrease the value of the `FRAMES_PER_SECOND` variable, this would lead to a larger delays when updating the patterns.

WarDrake, 2020-05-27:
Been playing around with the library today

//Serial.setDebugOutput(true);

This removed the pm error, I am not sending it on a pull request because I'm not sure if the visualizer uses it in some way.

but the pm message itself is no longer showing up once this line is commented, leading to no reboots either.

WarDrake, 2020-07-04:
This is solved on the newest builds as long as you comment line 20.

closed 2020-07-04 by WarDrake