NimmLor/esp8266-fastled-iot-webserver #45: Home Assistant Integration does not work

papperone, 2020-05-21:
Hello, I've been trying multiple times but there's something wrong into the Home Assistant integration.
I activated the "auto-discovery" features via MQTT, all seems fine, lamp di discovered, light entity created with all fetaures (on/off, brightness, color, and effects list) but from Home Assistant only the on/off works :(
I cannot change brightness (I move the slide but no reaction) and I cannot change the current effect; even if I put the lamp in "Solid Color" from web server I cannot change the color from the Home Assistant color wheel.
Can you please help to understand if it's my fault or the auto-discovery is somehow broken?

thanks!

NimmLor, 2020-05-22:
Some MQTT issues were fixed in the master branch, are you using the release version or a clone of the master branch?

papperone, 2020-05-22:
I just downloaded the master branch 3 days ago, and I've the problem I've described...
Is there a newer version I could try?

WarDrake, 2020-05-23:
Hey, the lights works normally for power / color / brightness / effects when it comes to the default method for controlling them, if you want specific things like animations, automation, speed control or the like then you need to enable it through scripts and helpers that you write and send MQTT commands throuhg, I'll write a tutorial on that in a couple days with some examples of my automations

Here's a screenshot of my control interface for them, and all features work without an Issue, the best way to check them tho is by opening the webpage as well as the home assistant control, change the brightness on home assistant and check on the webserver if it changed there too.

If you have the most recent version of the master branch, the change should be automatic but if you don't see it, try refreshing the webpage.

if you see the brightness sync between those two interfaces, then it's probably your powerbrick that's giving you an Issue

[image]

WarDrake, 2020-05-23:
Can you send me all your changes to the library to run it on your particular build? I'll try to reproduce the Issue.

papperone, 2020-05-24:
> 
> 
> Can you send me all your changes to the library to run it on your particular build? I'll try to reproduce the Issue.

I did not change any library, just enter the parameter of my device (the twisted lamp from thingiverse).
From Home Assistant I can turn on/off but brightness does not work (I change brighntess and nothign happens, but if I change from the web page the HA updates correctly).
About effects and color, when I open the light entity I can see effect list, but if I select one of them it does not change in the lamp nor in the web-page, if I do it in the web-page after some seconds in HA I can see it changing
tell me what you need to do your test and I will send it to you

WarDrake, 2020-05-25:
From you comment it seems HA is not sending the information to the correct topic, might be a config issue, but if you didn't change the config I might have made a mistake on there somewhere that I didn't see.

I did the whole MQTT/HA integration using the Nanoleafs, so there might be something on the lamp that is different and I failed to see, I'll double check that and get back at you, in the meantime it'd help me if you can copy the file exactly as you uploaded it to your device and put it on pastebin so I can look at it and test with it, it's probably something minor that I overlooked.

WarDrake, 2020-05-27:
@papperone apparently the incoming buffer was being overwritten with the outgoing one, there's a pull request with the fix now, once it's been accepted can you please update your device and report back?  
Thanks.

papperone, 2020-05-27:
ok I will do it but... what ius the best way to update my current device? 
as all the config I did is in the code, when I will download the new one it will be again empty and I need to do it all from scratch?
wouldn't be nice to have all configs in a separate file, maybe saved to SPIFF, so firmware will be more generic and easier to upgrade?
I have just a lamp done, building 2 more and I don't find an easy way to manage code rather than duplicate the whole folder and edit the config

WarDrake, 2020-05-27:
If you did a git clone of the current code, doing a git pull will update the software AND keep your changes, if you just downloaded it, then yes, you have to do any changes you did before again.
your secrets.h file isn't going anywhere tho, and the changes on the ino file are minimal, but since this is basically firmware, you have to do the changes again for each device.  

Nimmlor has made an incredible amount of work to make this almost plug and play so far tho, it could be a LOT of work if it weren't for how much thought he's put into this.

tech-freak, 2020-05-27:
> Hey, the lights works normally for power / color / brightness / effects when it comes to the...

sorry for off-topic but what theme is it ?

WarDrake, 2020-05-27:
@tech-freak It's iOS Dark Mode Theme with a custom background http://eskipaper.com/images/dark-gradient-wallpaper-1.jpg

WarDrake, 2020-06-09:
@papperone the fix has been merged into this repo, if you can test and confirm it's working as intended now, that'd be great, thanks.

papperone, 2020-06-09:
hi, sorry for the delay, I've been busy with my job...
I will test it as soon as possible and provide you feedback, thanks!

jhopkins213, 2020-06-13:
Downloading the master worked perfect for me Thank You

NimmLor, 2020-06-16:
Added fixes to (Release 4.1)(https://github.com/NimmLor/esp8266-fastled-iot-webserver/releases/tag/4.1)

closed 2020-06-16 by NimmLor