NimmLor/esp8266-fastled-iot-webserver #47: between patterns it flash short to white (flash)

hexxter, 2020-05-28:
HW: MCU Wemos Mini without level converter (but i see no artifacts i think this ist not the problem)
it flash if it change the pattern its not good to see if the brightness is high but with low brightness. I will use it as night lamp for my childs but if its flash i can it not use.

br Hexxter

MaxHuerlimann, 2020-05-31:
I am getting the same problem, using the wemos d1 mini pro. Also not using a level converter without seeing any artifacts. Running just 24 ws2812b leds

NimmLor, 2020-05-31:
I haven't figured out yet why the wemos crashes but removing the following line from the code should fix it:

`#define FASTLED_ALLOW_INTERRUPTS 0`

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo