NimmLor/esp8266-fastled-iot-webserver #49: MQTT connection unstable

iammenotyou, 2020-06-08:
Using the latest main branch I noticed after a period of time I was not able to communicate with the leafs.  So I attached to usb/serial monitor and noticed it would crash occasionally.

`12:10:05.520 -> Heap: 35488
12:10:05.520 -> Boot Vers: 31
12:10:05.520 -> CPU: 80
12:10:05.520 -> SDK: 2.2.2-dev(38a443e)
12:10:05.520 -> Chip ID: 5195897
12:10:05.520 -> Flash ID: 1454108
12:10:05.520 -> Flash Size: 4194304
12:10:05.520 -> Vcc: 65535
12:10:05.520 -> 
12:10:05.520 -> scandone
12:10:05.520 -> state: 0 -> 2 (b0)
12:10:05.555 -> state: 2 -> 3 (0)
12:10:05.555 -> SPIFFS contents:
12:10:05.555 -> state: 3 -> 5 (10)
12:10:05.555 -> add 0
12:10:05.555 -> aid 1
12:10:05.555 -> cnt 
12:10:05.590 -> 
12:10:05.590 -> Connecting to Wifi
12:10:05.590 -> 
12:10:05.590 -> connected with Wifi, channel 6
12:10:05.590 -> dhcp client start...
12:10:05.624 -> HTTP web server started
12:10:06.631 -> ip:192.168.0.251,mask:255.255.0.0,gw:192.168.0.1
12:10:15.554 -> pm open,type:0 0
12:10:35.575 -> pm 1990
12:10:37.573 -> 
12:10:37.573 -> Soft WDT reset
12:10:37.573 -> 
12:10:37.573 -> >>>stack>>>
12:10:37.573 -> 
12:10:37.573 -> ctx: sys
12:10:37.573 -> sp: 3fffed40 end: 3fffffb0 offset: 01b0
12:10:37.573 -> 3fffeef0:  4025c4ac 01ccd177 60000600 4022cc91  
12:10:37.573 -> 3fffef00:  00000000 3ffee4b8 00000000 00000000  
12:10:37.607 -> 3fffef10:  4024c1ec 3ffee4b8 3ffef470 60000600  
12:10:37.607 -> 3fffef20:  4024eb65 3ffee4b8 3ffef470 01cd49f4  
........
12:10:38.774 -> 3fffffa0:  3fffdad0 00000000 3fff11d8 40218f45  
12:10:38.774 -> <<<stack<<<
12:10:38.808 -> 
12:10:38.808 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
12:10:38.808 -> 
12:10:38.808 -> load 0x4010f000, len 1392, room 16 
12:10:38.808 -> tail 0
12:10:38.808 -> chksum 0xd0
12:10:38.808 -> csum 0xd0
12:10:38.808 -> v3d128e5c
12:10:38.808 -> ~ld
12:10:38.891 -> ⸮⸮
12:10:38.993 -> Heap: 35488
12:10:38.993 -> Boot Vers: 31
12:10:38.993 -> CPU: 80
12:10:38.993 -> SDK: 2.2.2-dev(38a443e)
12:10:38.993 -> Chip ID: 5195897
12:10:38.993 -> Flash ID: 1454108
12:10:38.993 -> Flash Size: 4194304
12:10:38.993 -> Vcc: 65535
12:10:39.028 -> 
12:10:39.028 -> scandone
12:10:39.028 -> state: 0 -> 2 (b0)
12:10:39.028 -> state: 2 -> 3 (0)
12:10:39.028 -> SPIFFS contents:
12:10:39.028 -> state: 3 -> 5 (10)
12:10:39.028 -> add 0
12:10:39.028 -> aid 1
12:10:39.028 -> cnt 
12:10:39.063 -> 
12:10:39.063 -> Connecting to Coruscant
12:10:39.097 -> 
12:10:39.097 -> connected with Coruscant, channel 6
12:10:39.097 -> dhcp client start...
12:10:39.097 -> HTTP web server started
12:10:39.132 -> ip:192.168.0.251,mask:255.255.0.0,gw:192.168.0.1
12:10:49.042 -> pm open,type:0 0`

So I did some further digging.   It seems when the code is first flashed to the Wemos D1 mini it connects to mqtt and communicates fine.  It appears, that after it is reset it will  not connect even attempt to communicate with the mqtt server.   To eliminate possible issues with the mqtt broker, I used mosquitto on a raspberry pi and RabbitMQ on a linux desktop with the same response.   I've also tried 3 wemos d1 mini's all with the same behavior.

In the code I changed the config and secrets.h file only.
`/*######################## MAIN CONFIG ########################*/
#define LED_TYPE		WS2812				// You might also use a WS2811 or any other strip that is Fastled compatible 
#define DATA_PIN		D7					// Be aware: the pin mapping might be different on boards like the NodeMCU
//#define CLK_PIN		D5				// Only required when using 4-pin SPI-based LEDs
#define CORRECTION		UncorrectedColor    // If colors are weird use TypicalLEDStrip
#define COLOR_ORDER		GRB					// Change this if colors are swapped (in my case, red was swapped with green)
#define MILLI_AMPS		18000				// IMPORTANT: set the max milli-Amps of your power supply (4A = 4000mA)
#define VOLTS			5					// Voltage of the Power Supply

//#define REMOVE_VISUALIZATION		// remove the comment to completly disable all udp-based visualization patterns

#define HOSTNAME "Nanoleaf"				// Name that appears in your network, don't use whitespaces, use "-" instead

#define DEVICE_TYPE 4				// The following types are available
`

`#elif DEVICE_TYPE == 4			// Nanoleafs
	#define LEAFCOUNT 12              // Amount of triangles
	#define PIXELS_PER_LEAF 38        // Amount of LEDs inside 1x Tringle
`

`	#define ENABLE_OTA_SUPPORT				// requires ArduinoOTA - library, not working on esp's with 1MB memory (esp-01, Wemos D1 lite ...)
		#define OTA_PASSWORD "password1"	//  password that is required to update the esp's firmware wireless
`

`	#define ENABLE_MQTT_SUPPORT			// allows integration in homeassistant/googlehome/mqtt, 
`

`  const char* mqttServer = "192.168.0.10";
	const int mqttPort = 1883;
`

`// AP mode password
 const char WiFiAPPSK[] = "yourpasswordhere1";

// Wi-Fi network to connect to (if not in AP mode)
 char* ssid = "wifi";
 char* password = "password";

// only required when MQTT is enabled
const char* mqttUser = "mqtt-user";
const char* mqttPassword = "mqttpassword";`

Any help would be appreciated.  Love the project!

WarDrake, 2020-06-08:
The current branch has a bug in the MQTT integration where buffers get overwritten, most likely that's the Issue you're seeing, there is a fix already done and in a pull request, it will fix this issue once it's merged, you can test the master branch on my fork if you're in a hurry.

if not that, another thing I've noticed sometimes happens, is that it seems settings get stuck, if you make any change on the webpage they seem to release and mqtt starts working again, that seems to be a different issue tho, did you check that?

and a last one, is that pm open, type:0 0 message you see there, it goes away if you comment this line :  #define FASTLED_ALLOW_INTERRUPTS 0
that also solved another instability Issue for me.

iammenotyou, 2020-06-09:
Since it was only a couple of lines, I copied over the multicast code changes & commented out the FASTLED_ALLOW.   18 hours no crashes and mqtt still communicating.   Thanks for all your help!

closed 2020-06-09 by iammenotyou