NimmLor/esp8266-fastled-iot-webserver #50: [HELP] Server's changes are not applied to ledstrip

therealmrcolor, 2020-06-10:
The first time that I installed server and upload it on the D1 worked prefectly: I went to the server page and I changed the pattern and the color and all worked (it was report on the arudino serial monitor). At the time the led strip was not connected to the board beacuse I wasn't sure about the wire, so I only tried the "code's part". The next day leared about how to connect the wire I upload the server again but it won't work anymore; after many tries I decide to switch to another pc, but the result will be the same:
-I upload the program on the board, then I do the esp8266 sketch upload 
-I open the serial monitor and i push the reset button on the board
-the board connects to my wifi and got me his ip
-I paste the on my browser and I reached up the server page 
The problem starts here, when I, for example, change the pattern, the indicator on the bottom confirm the change but in the serial monitor nothing is happening.

Information that could be useful:
-my led strip WS2811 (12volt alimented by an external supply)
-i'm running windows 10 with arduino ide 1.8.12,  the library are the same you used
-i'm using a samsung and a hauwei usb cable to connect to my pc to my wedos d1

I'm attaching you a video so you can better understand
[https://www.youtube.com/watch?v=74ELmy13PaY](url)

WarDrake, 2020-06-12:
I don't think the most recent version will print out anything to the serial about a pattern change, but I do notice you're not using a resistor on your data connection, try putting a 1k resistor in that data line and make sure the pins make good contact.

bb-Ricardo, 2021-01-17:
Hi, you still experience any issues?

You can try the latest "in-development" branch and set the LED_DEBUG to `1`. There you can see if the changes are received by the esp. If so, then it is very likely an issue with your LED or PIN setup.

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo