NimmLor/esp8266-fastled-iot-webserver #53: First leaf randomly flickers when off

flashez, 2020-06-17:
When my nanoleafs are off, the first leaf randomly turns on for a fraction of a second (blue color) for no reason. I would say it happens roughly every 5 minutes.
This wasn't happening with the old firmware (https://github.com/NimmLor/esp8266-nanoleaf-webserver).

WarDrake, 2020-06-17:
That means You've likely have a reboot issue, as booting will cause that, connect it to a serial monitor and keep a look on it for an hour or so, the error there should help diagnose why it's rebooting.

flashez, 2020-06-17:
> That means You've likely have a reboot issue, as booting will cause that, connect it to a serial monitor and keep a look on it for an hour or so, the error there should help diagnose why it's rebooting.

Updated to today's update, will see.

ZeppDK, 2020-07-06:
damit, i have the same issue after updating nanoleaf to the latest version... 
i'll see if i can get a log enough cable to hook it up over some time :)

ZeppDK, 2020-07-06:
what baudrate shout it have?
im getting nonsens right now, and it looks like when connecting to a router with the wrong baud rate...

ZeppDK, 2020-07-06:
nevermind... got it working at 115200...
collecting data :)

ZeppDK, 2020-07-06:
[nanoleaf_crash.txt](https://github.com/NimmLor/esp8266-fastled-iot-webserver/files/4880258/nanoleaf_crash.txt)

there is not alot of data when its crashing, but maybee you can use this to something @WarDrake

WarDrake, 2020-07-06:
are you using a weemos D1 or a nodeMCU?
or some other board?

ZeppDK, 2020-07-06:
WeMos D1 Mini

WarDrake, 2020-07-06:
try applying this patch :
https://github.com/WarDrake/esp8266-nanoleaf-webserver/commit/e3e5124d1d7de959cfe474fa6ecfa506c967b5a3

ZeppDK, 2020-07-06:
with that mod, it has been running for 2 hours without rebooting, so it seems to have fixed the error.

bb-Ricardo, 2021-01-17:
Closing this issue. Please reopen if the issue persists.

closed 2021-01-17 by bb-Ricardo