NimmLor/esp8266-fastled-iot-webserver #55: Can't configure more than 4 Lines of 10 LEDS in DeskLamp

mrvanes, 2020-07-02:
I've successfully managed to compile and upload the sketch for DEVICE_TYPE 3 using LINE_COUNT 4 and LEDS_PER_LINE 10, but as soon as I rais LINE_COUNT to 5, I have an exception on the ESP8266EX:

It very much feels like an out-of-memory issue, but it looks like your example DeskLamp also contains 8x10 = 80 LEDS and are successfully managed by the ESP?

```
Heap: 39016
Boot Vers: 31
CPU: 80
SDK: 2.2.2-dev(38a443e)
Chip ID: 3220022
Flash ID: 1458208
Flash Size: 4194304
Vcc: 65535

scandone
no Olympus-home found, reconnect after 1s
SPIFFS contents:
FS File: /js/bootstrap.min.js, size: 37045
FS File: /js/app.js, size: 12933
FS File: /js/r-websocket.min.js, size: 3100
FS File: /js/jquery-3.1.1.min.js, size: 86709
FS File: /js/minicolors.min.js, size: 15463
FS File: /js/dark-mode-switch.min.js, size: 492
FS File: /favicon.ico, size: 4286
FS File: /fonts/glyphicons.woff2, size: 18028
FS File: /index.htm, size: 9959
FS File: /css/styles.css, size: 2152
FS File: /css/bootstrap.min.css, size: 121093
FS File: /css/minicolors.min.css, size: 96597
FS File: /css/dark-mode.css, size: 1104
FS File: /images/github.ico, size: 6518
FS File: /images/atom196.png, size: 3889
FS File: /images/surr.png, size: 5322

Connecting to Olympus-home
HTTP web server started
reconnect
Fatal exception 28(LoadProhibitedCause):
epc1=0x4020b16b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000027, depc=0x00000000

Exception (28):
epc1=0x4020b16b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000027 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 0190
3fffff00:  1a1a0000 0002031a 00000000 3fff0254  
3fffff10:  0000001a 3fff0260 3fff0114 4020b575  
3fffff20:  4021e910 00000000 00001388 3fff0520  
3fffff30:  00000000 00000001 4bc6a7f0 00000000  
3fffff40:  00000000 00000000 3fff0114 402074d2  
3fffff50:  00000000 00000008 3fff0260 402148ba  
3fffff60:  007a1200 1a8dc60e 3fff0200 40209028  
3fffff70:  3fffdad0 00000000 3fff04e0 402077c1  
3fffff80:  00000000 00000000 00000001 40100598  
3fffff90:  3fffdad0 00000000 3fff04e0 401005b9  
3fffffa0:  3fffdad0 00000000 3fff04e0 4021499c  
3fffffb0:  feefeffe feefeffe 3ffe8510 401010f5  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
```

The stack exception decoder decodes the stack to:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4020b16b: WiFiClient::WiFiClient(ClientContext*) at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/include/ClientContext.h line 112
EXCVADDR: 0x00000027

Decoding stack results
0x4020b575: WiFiServer::available(unsigned char*) at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/WiFiServer.cpp line 125
0x402074d2: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h line 306
0x402148ba: __yield() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_main.cpp line 112
0x40209028: CFastLED::delay(unsigned long) at /home/martin/Arduino/libraries/FastLED/FastLED.cpp line 134
0x402077c1: loop() at /home/martin/Projects/DeskLamp/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino line 1122
0x40100598: ets_post(uint8, ETSSignal, ETSParam) at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_main.cpp line 177
0x401005b9: esp_schedule() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_main.cpp line 125
0x4021499c: loop_wrapper() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/core_esp8266_main.cpp line 197
```

This is my Secrets.h
```
// AP mode password
const char WiFiAPPSK[] = "your-password";

// Wi-Fi network to connect to (if not in AP mode)
char* ssid = "Olympus-home";
char* password = "******";

// only required when MQTT is enabled
const char* mqttUser = "test";
const char* mqttPassword = "test";

#define LED_TYPE      WS2813
#define DATA_PIN      D1
//#define CLK_PIN     D2
//#define CORRECTION  TypicalLEDStrip
//#define COLOR_ORDER GRB
#define MILLI_AMPS    1500
#define VOLTS         5
#define HOSTNAME      "desklamp"

#define DEVICE_TYPE   3
#define LINE_COUNT    5 <-- This fails, with 4 it boots correctly
#define LEDS_PER_LINE 10

//#define ACCESS_POINT_MODE
#define ENABLE_OTA_SUPPORT
#define ENABLE_MULTICAST_DNS
//#define RANDOM_AUTOPLAY_PATTERN
//#define AUTOPLAY_IGNORE_UDP_PATTERNS
//#define ENABLE_ALEXA_SUPPORT
//#define SOUND_SENSOR_SUPPORT
//#define ENABLE_SERIAL_AMBILIGHT
//#define ENABLE_MQTT_SUPPORT
#define REMOVE_VISUALIZATION
```

Any hints on what I can try to get the 80 LEDs working?

mrvanes, 2020-07-02:
DEVICE_TYPE 0 with NUM_LEDS 80 and BAND_GROUPING 1 works flawless, so the 80 LEDS is not the problem, maybe the math for the DeskLamp is somewhat hungry?

mrvanes, 2020-07-02:
I've boiled it down to any one of the first 5 DEVICE_TYPE 3 patterns. Each one of them triggers the exception and the problem is that once that's stored in eeprom, you're in an endless loop, because that pattern will be retrieved on start. If I replace line 1259 with 
```currentPatternIndex = 5;```
Everything works like expected (apart from remembering the last pattern of course), except that I can't use the DeskLamp patterns.

NimmLor, 2020-07-02:
I have now idea why it wouldn't boot, try removing the `FASTLED_ALLOW_INTERRUPTS` line at the top. Then update all libraries to the newest version and a screenshot of the settings in the *tools* menu would be great too.

mrvanes, 2020-07-02:
My sincere apologies, I mistakenly assumed Secrets.h had to contain all configuration, so I had a strange mixed config that made no sense at all. I now corrected it to only contain the secrets and adjusted the top of the sketch to have the correct configuration and all is well now. Thanks for the beautiful project!

closed 2020-07-02 by mrvanes