NimmLor/esp8266-fastled-iot-webserver #57: AP mode not stable

mrvanes, 2020-07-06:
I'm trying to change the WiFi settings from client to AP mode, but the AP is unstable. My laptop disconnects within 5 seconds, telling me that the SSID disappeared in syslog. The AP actually disappears until the device is rebooted. There's a lot of AP bugs filed on esp8266/arduino github so it seems I'm not the only one. Has anyone successfully connected to this fastled server using AP on an esp8266 12E (Wemo D1 lookalike)?

These are the only lines in serial output:
```
add 1
aid 1
station: 00:28:f8:c1:3a:54 join, AID = 1
wifi evt: 5
wifi evt: 9
wifi evt: 7
wifi evt: 7
station: 00:28:f8:c1:3a:54 leave, AID = 1
rm 1
```

DrJMKatz, 2020-07-18:
Same issue here, though I got a few more errors:
```Connect to Wi-Fi access point: ESP8266 Thing 8553
and open http://192.168.4.1 in your browser
HTTP web server started
add 1
aid 1
station: 10:cd:b6:14:71:ce join, AID = 1
station: 10:cd:b6:14:71:ce leave, AID = 1
rm 1
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 9(StoreProhibitedCause):
Fatal exception Fatal exception 
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
```

bb-Ricardo, 2021-01-17:
Currently AP mode is not supported anymore.

If you need it I would kindly ask you to open a feature request. Thank you.

closed 2021-01-17 by bb-Ricardo