NimmLor/esp8266-fastled-iot-webserver #59: Feature Request: Physical Button support

rossiniscarface, 2020-07-07:
Hi,

very nice project and feature set. 
Would it be possible to add some functions for physical buttons/switches.
e.g.: 
long button press —> switch device on/off
short button press —> change color 
double short button press —> change brightness.

I know everybody uses alexa and mqtt and what not. But i think for some people the possibility to switch a light on and off directly via button or other physical input is nice. Also it would allow some degree of indepentent control if the infrastructure not working (wifi down/smartphone dead/mqtt server down ...)

thanks

bb-Ricardo, 2020-12-12:
found this here: https://github.com/Hotaman/esp8266-fastled-iot-webserver/commit/c797b23250f861cd97c7742d9bb2ffb2650ffe9b

bb-Ricardo, 2021-01-17:
Still interested in this?

Short and long press seem doable. But not sure about brightness. I would think a potentiometer would way better suited.

rossiniscarface, 2021-01-18:
Still interested in this. 
I did just some suggestions, but i think a potentiometer is overkill. Looking at a lot of lamps there are very often somerhing like this:
first tap on 25% brightness
second 50%
third 75%
fourth 100%
fifth off 

Just on off would be ok for me too.

bb-Ricardo, 2021-01-18:
This sounds great.

Following sugestion:
Initial state: off
 1st short press: 100% / on
 2nd short press: 50%
 3rd short press: 20%
 4th short press: off

 Long press: next pettern

What do you think?