NimmLor/esp8266-fastled-iot-webserver #62: [Feature Request/Enhancement] toggle for default behavior on boot

cegan09, 2020-07-16:
Was wondering if it would be possible to have a way to change default behavior on boot. Right now it resumes last mode, which is great for most applications. I'm going to be using this project in a light fixture retrofit where 90% of the time it will simply be on solid white as a normal light. In this corner case I'd want the default behavior when power is removed and then applied again to be to always default back to solid color mode so I don't have to change something in the web interface when reverting back to normal light use from the colorful modes. Was envisioning a toggle in the web interface to have default boot mode switch between "last set" and "fixed" with the option to pick the fixed behavior. 

I believe I can figure setting this behavior in the code for my one off application, I just didn't know if the idea resonated with your development. If it's not something you think is worth adding no worries.

NimmLor, 2020-07-27:
Hi, I probably won't implement this because it's too much of a corner case, but it's easily implemented: 

just replace this stuff in the `loadsettings()` function:
```
    brightness = EEPROM.read(0);

    currentPatternIndex = EEPROM.read(1);
    if (currentPatternIndex < 0)
        currentPatternIndex = 0;
    else if (currentPatternIndex >= patternCount)
        currentPatternIndex = patternCount - 1;

    byte r = EEPROM.read(2);
    byte g = EEPROM.read(3);
    byte b = EEPROM.read(4);
```
with this:
```
    brightness = 255;
    currentPatternIndex = patternCount - 1;
    byte r = 255;
    byte g = 255;
    byte b = 255;
```

bb-Ricardo, 2021-01-12:
Is this still an issue or could this be closed?

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo