NimmLor/esp8266-fastled-iot-webserver #63: [HELP] the code have something bad?

Guspin, 2020-07-21:
The "setSolidColor(d->getR(), d->getG(), d->getB());" which is on line 3611 is sending compilation errors:

`In file included from C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:23:0:
C:\Users\guspi\OneDrive\Documentos\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from C:\Users\guspi\OneDrive\Documentos\Arduino\libraries\FastLED/FastLED.h:65:0,
                 from C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:23:
C:\Users\guspi\OneDrive\Documentos\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void mainAlexaEvent(EspalexaDevice*)':
esp8266-fastled-iot-webserver:3610:54: error: no matching function for call to 'setSolidColor(uint8_t, uint8_t, uint8_t)'
         setSolidColor(d->getR(), d->getG(), d->getB());
                                                      ^
C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:3610:54: note: candidates are:
C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:628:6: note: void setSolidColor(uint8_t, uint8_t, uint8_t, bool)
 void setSolidColor(uint8_t r, uint8_t g, uint8_t b, bool updatePattern) {
      ^
C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:628:6: note:   candidate expects 4 arguments, 3 provided
C:\Users\guspi\Desktop\Proyectos actuales\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:641:6: note: void setSolidColor(CRGB, bool)
 void setSolidColor(CRGB color, bool updatePattern)
 
exit status 1
no matching function for call to 'setSolidColor(uint8_t, uint8_t, uint8_t)'`

I checked the versions of the libraries and nothing is wrong

NimmLor, 2020-07-21:
Hi, I've noticed that issue too, it is caused by a recent merge request. It should be fixed in the [in-development branch](https://github.com/NimmLor/esp8266-fastled-iot-webserver/tree/in-development). Clone the code from there and try again.

Sorry for the inconvenience.

Guspin, 2020-07-21:
Thanks for answering!

Guspin, 2020-07-21:
Thanks for answering! 
I spend some hours reading the code, and found out that what it fixed was "setSolidColor(d->getR(), d->getG(), d->getB(), **true**);"   that "true" was a few lines down of where the issue was happening it works on the in-dev code and the normal one, at least it compiles, I haven't try it yet to put it on to the board.

kroland705, 2020-09-08:
Hi, I have the same issue even after downloading from the in-dev fork. Here are the errors from the IDE: pastebin.com/Er3qwiXf

bb-Ricardo, 2021-01-12:
Can you try out the current master branch? It should be fixed by now.

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo