NimmLor/esp8266-fastled-iot-webserver #65: Alexa doesn't find the ESP

paul-arnold, 2020-07-25:
Hi,
I enabled Alexa Support by uncommenting #define ENABLE_ALEXA_SUPPORT and uncommented #define AddAutoplayDevice (...).
I also installed the needed library.
Using the WebUI, I navigate to the "Alexa Pairing" tab which gives me a new screen, telling me to search for devices.
I'm using the Alexa app on my iPhone for that (ESP + Phone are in the same WiFi).
However, the Alexa app doesn't find any new devices. 

I went through the code but honestly, I don't find a starting point to debug this issue. 

Any help is appreciated.

NimmLor, 2020-07-27:
Is your Echo device located in the same network as the esp?
What echo device and which generation of it are you using.

You can add `#define ESPALEXA_DEBUG` before `#include <Espalexa.h>` to get information in the serial monitor when attempting to pair the device.

paul-arnold, 2020-07-27:
Yes, I'm tried the iPhone app and the browser to discover from within the same Wifi. 
There's no physical Echo device present at this time.
I already tried adding ESPALEXA_DEBUG, I couldnt find anything special in the output.
There were UDP requests from a few devices but nothing the Wemos did respond to.

NimmLor, 2020-07-27:
Unfortunatly the library states: 

> Espalexa only works with a genuine Echo speaker, it probably wont work with Echo emulators, RPi homebrew devices or just the standalone app.

I don't think that I can do anything for you :(

closed 2020-07-27 by paul-arnold

emersonsc, 2022-01-12:
> I read that in the lib description after creating the issue, too :) I'll close the issue for now and order a 3rd gen Echo to test it

Any luck? My echo doesnt find mine either

Roycinger, 2022-01-13:
> > I read that in the lib description after creating the issue, too :) I'll close the issue for now and order a 3rd gen Echo to test it
> 
> Any luck? My echo doesnt find mine either

With the 3rd Gen Echo, I got it to work.
I'm facing another issue with the sketch but the creator seems not to answer issues unfortunately