NimmLor/esp8266-fastled-iot-webserver #67: Improve webpage

garethbradley, 2020-07-26:
Hey,

I was wondering about rejigging the webpage? Couple of suggestions:

- Grouping settings by adding a "category" to the Fields.h, and change to JSON function in Field.h
- Only show relevant settings by pattern by updating the patterns (PatternAndName struct too) to include the relevant fields
- Some king of graphical nanoleaf designer would be awesome 😃 🤞 

Again, keep up the great work and if you like the ideas, I'm happy to jump in a do a PR.

Cheers
Gareth

NimmLor, 2020-07-27:
Hi, thanks for your feedback, I'll keep some things in mind for the next release.
Unfortunatly due to the small storage space (512kb/1024kb) of the esp it's pretty hard to add complex features, because I want to avoid linking in stylesheets/libraries etc. from the internet. With bootstrap, jquere and mincolors we've already got around 450kb which leaves very little space for everything else.

bb-Ricardo, 2021-01-12:
Current master branch has some improvements for the web interface. Can you give them a try and let me know what you think?

Thank you.

bb-Ricardo, 2021-02-09:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-02-09 by bb-Ricardo