NimmLor/esp8266-fastled-iot-webserver #68: Wemos D1 keeps restarting

paul-arnold, 2020-07-27:
Hello,
I successfully manages to get 1 leaf running.
Apart from a few quirks (Wemos did not connect to Wifi everytime it booted up) it was running fine and I was able to adjust patterns and colors.
On on point, the Wemos decided to not work anymore. 
Looking at the serial monitor, I saw that it is constantly rebooting.
So I tried another Wemos, unfortunately with the same result (keeps rebooting).
I tried another simple sketch (blink) to verify the Wemos(es) aren't broken, this works fine without reboots.
So I thought I messed something up in the code and discarded all changes, downloaded the github archive again.
Only things I adjusted are the secrets.h and the number of leafs.

How can I debug further?
Thanks for assistance!

axiscpi, 2020-08-13:
Hello,
It seems I have the same problem, after a few minutes using the web app, the wemos restarts indefinitely. The only way to bring back working, is to reinstall the arduino sketch with option "Erase Flash = All flash content".

Thanks for your help

RosalieWessels, 2020-09-27:
Hi, 

I have the exact same problem with an Esp8266. I also noticed that the problem started when the rainbow dots pattern was used. Erasing all flash removing the rainbow dots pattern like @axiscpi mentioned seemed to have solved the problem (for now).

Roycinger, 2020-09-27:
Yep, I tried that too. Seems to be an issue with the rainbow pattern.

bb-Ricardo, 2021-01-17:
Hi,

can you check out the current "in-development" branch and switch LED_DEBUG to `1`? Please let us know if the issue still persists.

Thank you.

bb-Ricardo, 2021-02-09:
Tested the setting and had no issues with current 'in-development' branch.

Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-02-09 by bb-Ricardo