NimmLor/esp8266-fastled-iot-webserver #70: Fire effect proporsal

KrChm, 2020-07-28:
I found realy nice fire efect
Best settings: 
-speed 18
-cooling 20
-sparking 120

#define MAXHOT sparking*HEIGHT

void heatMap(CRGBPalette16 palette, bool up)
{
    fill_solid(leds, NUM_LEDS, CRGB::Black);

    // Add entropy to random number generator; we use a lot of it.
    random16_add_entropy(random(256));

     static unsigned int spark[LENGTH]; // base heat
  CRGB stack[LENGTH][HEIGHT];        // stacks that are cooler

  // 1. Generate sparks to re-heat
  for( int i = 0; i < LENGTH; i++) {
    if (spark[i] < sparking ) {
      int base = sparking * 2;
      spark[i] = random16( base, MAXHOT );
    }
  }

  for( int i = 0; i < LENGTH; i++) {
    spark[i] = qsub8( spark[i], random8(0, cooling) );
  }
  for( int i = 0; i < LENGTH; i++) {
    unsigned int heat = constrain(spark[i], sparking/2, MAXHOT);
    for( int j = HEIGHT-1; j >= 0; j--) {
      byte index = constrain(heat, 0, sparking);
      stack[i][j] = ColorFromPalette(palette, index);
      unsigned int drop = random8(0,sparking);
      if (drop > heat) heat = 1;
      else heat -= drop;
      heat = constrain(heat, 0, MAXHOT);
    }
  }
  for( int i = 0; i < LENGTH; i++) {
  for( int j = 0; j < HEIGHT; j++) {
     leds[(j*LENGTH) + i] = stack[i][j];
  }
  }

  FastLED.delay(1000/speed);
}

bb-Ricardo, 2021-01-09:
isn't this already included? https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/esp8266-fastled-iot-webserver.ino#L1853

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo