NimmLor/esp8266-fastled-iot-webserver #74: Not able to compile when enabling MQTT

Diondk, 2020-10-07:
Hi, 

when i enable mqtt support and provide the information, when i want to upload i get the following errors:

`C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void loop()':
esp8266-fastled-iot-webserver:1170:28: error: 'class PubSubClient' has no member named 'setKeepAlive'
                 mqttClient.setKeepAlive(10);
                            ^
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void mqttCallback(char*, byte*, unsigned int)':
esp8266-fastled-iot-webserver:4259:37: error: no matching function for call to 'setSolidColor(int&, int&, int&)'
             setSolidColor(cr, cg, cb);
                                     ^
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:4259:37: note: candidates are:
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:630:6: note: void setSolidColor(uint8_t, uint8_t, uint8_t, bool)
 void setSolidColor(uint8_t r, uint8_t g, uint8_t b, bool updatePattern)
      ^
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:630:6: note:   candidate expects 4 arguments, 3 provided
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:644:6: note: void setSolidColor(CRGB, bool)
 void setSolidColor(CRGB color, bool updatePattern)
      ^
C:\Users\310038157\Downloads\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:644:6: note:   candidate expects 2 arguments, 3 provided
exit status 1
'class PubSubClient' has no member named 'setKeepAlive'
`

when i disable mqtt i am able to upload just fine. 

Kind Regards,
Dion

WarDrake, 2020-10-07:
You migth have an old version of the pubsubclient library, check that, if that doesn't solve the problem, 
go to line : 1172   mqttClient.setKeepAlive(10);
and edit it to be : mqttClient::setKeepAlive(10);
or just comment it out if that doesn't work, the timeout issue was solved some time ago and this line is no longer strictly necesary.

Diondk, 2020-10-07:
OK pubsubclient is version 2.8.0 the mqttcallback issue is now gone. Only have the other issues left, Where the void setSolodColr appears:
`esp8266-fastled-iot-webserver.ino: In function 'void mqttCallback(char*, byte*, unsigned int)':
esp8266-fastled-iot-webserver:4261:37: error: no matching function for call to 'setSolidColor(int&, int&, int&)'
             setSolidColor(cr, cg, cb);
                                     ^
esp8266-fastled-iot-webserver.ino:4261:37: note: candidates are:
esp8266-fastled-iot-webserver.ino:632:6: note: void setSolidColor(uint8_t, uint8_t, uint8_t, bool)
 void setSolidColor(uint8_t r, uint8_t g, uint8_t b, bool updatePattern)
      ^
esp8266-fastled-iot-webserver.ino:632:6: note:   candidate expects 4 arguments, 3 provided
esp8266-fastled-iot-webserver.ino:646:6: note: void setSolidColor(CRGB, bool)
 void setSolidColor(CRGB color, bool updatePattern)
      ^
esp8266-fastled-iot-webserver.ino:646:6: note:   candidate expects 2 arguments, 3 provided
Multiple libraries were found for "ArduinoJson.h"

exit status 1
no matching function for call to 'setSolidColor(int&, int&, int&)'`

WarDrake, 2020-10-07:
Edit line 4261
from :
setSolidColor(cr, cg, cb);
to :
setSolidColor(cr.toInt(), cg.toInt(), cb.toInt(), false);

Diondk, 2020-10-07:
Ok something seems to be wrong with my arduino setup changed that line still get an error.

`
esp8266-fastled-iot-webserver:4261:30: error: request for member 'toInt' in 'cr', which is of non-class type 'int'
             setSolidColor(cr.toInt(), cg.toInt(), cb.toInt(), false);
esp8266-fastled-iot-webserver:4261:42: error: request for member 'toInt' in 'cg', which is of non-class type 'int'
             setSolidColor(cr.toInt(), cg.toInt(), cb.toInt(), false);
esp8266-fastled-iot-webserver:4261:54: error: request for member 'toInt' in 'cb', which is of non-class type 'int'
             setSolidColor(cr.toInt(), cg.toInt(), cb.toInt(), false);

`

WarDrake, 2020-10-08:
I just updated the code to the latest version on master and compiled and uploaded the code without any Issues, I think you might be selecting the wrong development board, or something to that effect.

Check all your libraries are updated, the config of your arduino editor, and double check any changes you made to the script because I can't find any issues.

and if you keep getting issues with that setSolidColor function try setting line 4261 to 
setSolidColor(cr, cg, cb, false);
tho it shouldn't be necesary as it compiles fine over here...

Windows 10, 
Arduino Ide / Visual Code, 
FastLED 3.3.3
PubSubClient 2.8.0,
ArduinoJSON 6.16.1
ESP8266 2.7.1 (board manager)

exostasis, 2020-10-27:
> Edit line 4261
> from :
> setSolidColor(cr, cg, cb);
> to :
> setSolidColor(cr.toInt(), cg.toInt(), cb.toInt(), false);

Are you sure to set it to false wouldn't it be true since the original code when you updated MQTT used true as the default value? I personally just updated the signature for the method to default to true again

dchwilk, 2020-12-06:
I have the same problem

WarDrake, 2021-01-05:
This Issue has been fixed, can you guys confirm please?

The-Holgi, 2021-01-14:
Hello,
after comment out mqttClient.setKeepAlive(10) mqtt works for me.
How shows the syntax of payload?
I want to serve the led‘s over my fhem server

WarDrake, 2021-01-14:
If commenting the keepalive line fixes it for you, you're using an old version of the pubsub library,  try updating it.

The-Holgi, 2021-01-14:
Ok, thanks.
But the problem with serve over fhem is not solved. The state of the lights can i see in fhem but not more.

WarDrake, 2021-01-14:
The original Issue here is solved, please create a new Issue to ask for the payload syntax

closed 2021-01-14 by WarDrake

pattyland, 2021-01-31:
According to https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md you need to install PubSubClient, version 2.7.0. Is this info outdated and you need to install 2.8.0? I get this error too with 2.7.0 and version 4.4 of the Webserver:

```
Arduino: 1.8.13 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

'class PubSubClient' has no member named 'setKeepAlive'
```

bb-Ricardo, 2021-01-31:
Hi Sören,

did you try the latest "in-development" with the 2.8.0 PubSubClient Library? It should be fixed there.

Also need to update the sketch "data" with the uploader.

pattyland, 2021-02-01:
No, I didn’t. As the release notes of 4.4 states Audio Visualization is broken too but fixed in 4.5, I thought I should wait for this stable version. Or what do you think?

bb-Ricardo, 2021-02-01:
There is still quite some issues which should be handled before 4.5. But we try to keep the "in-development" branch working. 

https://github.com/NimmLor/esp8266-fastled-iot-webserver/milestone/1