NimmLor/esp8266-fastled-iot-webserver #76: WiFi Connection Issue

shieldsit, 2020-10-14:
WiFi only gets IP on first startup after flashing device.  After that it seems to hang and never gets an IP address.  I've tested on WeMOS D1, NodeMCU (2 types) and it's the same on both.  This is what the serial monitor shows:

_Heap: 38408
Boot Vers: 6
CPU: 80
SDK: 2.2.2-dev(38a443e)
Chip ID: 4480577
Flash ID: 1458270
Flash Size: 4194304
Vcc: 65535

SPIFFS contents:
FS File: /css/bootstrap.min.css, size: 121099
FS File: /css/dark-mode.css, size: 1104
FS File: /css/minicolors.min.css, size: 96597
FS File: /css/styles.css, size: 2243
FS File: /favicon.ico, size: 4286
FS File: /fonts/glyphicons.woff2, size: 18028
FS File: /images/atom196.png, size: 3889
FS File: /images/github.ico, size: 6518
FS File: /images/surr.png, size: 5322
FS File: /index.htm, size: 10194
FS File: /js/app.js, size: 13395
FS File: /js/bootstrap.min.js, size: 37051
FS File: /js/dark-mode-switch.min.js, size: 492
FS File: /js/jquery-3.1.1.min.js, size: 86713
FS File: /js/minicolors.min.js, size: 15473
FS File: /js/r-websocket.min.js, size: 3101

Connecting to MYSSID
HTTP web server started_

Any suggestions on what I can check?

Thanks!

WarDrake, 2020-10-14:
This happens before an internet connection is attempted, hold the console open for a while and see what errors you get from the connection attempt.

monkeydave, 2020-11-06:
I have been trying to connect to my generic ESP8266 module but having the same issue.
I can get the module to connect to my wifi network with a basic arduino sketch and also when the code is setup in access point mode. 

I can get this to work if I remove the if statement around the wifi.begin() function on line 771.

I am fairly new to arduino and I can't understand why String(WiFi.SSID() would already be the same as String(ssid)?

shieldsit, 2020-11-06:
If I remember correctly, my issue was the pin I chose for the LED control.  I can't remember exactly what pin I was using, I think D4 but not positive.  Once I moved it to the one that was in the code everything started working.

WarDrake, 2020-11-06:
@shieldsit is correct, depending on the board you use, the pin you configure might cause boot issues, try switching out the pin to a higher one

bb-Ricardo, 2021-01-09:
Can we consider this issue resolved?

shieldsit, 2021-01-09:
Yes, thank you!

closed 2021-01-12 by bb-Ricardo