NimmLor/esp8266-fastled-iot-webserver #77: ESP32 Support ?

guiguijke, 2020-10-24:
Hello ,
Not really an issue, but I was wondering if you'll support ESP32 in a near future ?
Regards

NimmLor, 2020-10-24:
No, unfortunatly not, I don't have that much time at the moment, sorry :/

bb-Ricardo, 2021-01-12:
I guess I need to get my hands on an ESP32.

bb-Ricardo, 2021-01-31:
you might wan to try the "feature-esp32-support" branch

enniob, 2021-02-04:
The wifi manager never start, when I look at the console I see the following

DEBUG [221001]: Trying to connect to Wifi
DEBUG [221002]: Stats: 120 frames/s, frame delay: 7500 us
DEBUG [222004]: Trying to connect to Wifi
DEBUG [222004]: Stats: 121 frames/s, frame delay: 7500 us

But I never see the access point to connect to.

bb-Ricardo, 2021-02-04:
Can you post the full serial monitor output right from the start?

Thank you.

enniob, 2021-02-04:
@bb-Ricardo here is the output from the start.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

DEBUG [209]: Loading config

DEBUG [221]: System Information:
DEBUG [221]: Heap: 205872
DEBUG [221]: SDK: master
DEBUG [221]: CPU Speed: 240 MHz
DEBUG [221]: Flash Size: 4194304KB
DEBUG [223]: MAC address: A4:CF:12:0B:0E:D4

E (375) SPIFFS: mount failed, -10025
An Error has occurred while mounting SPIFFS
DEBUG [242]: Setting: pattern: Solid Color
DEBUG [1007]: Trying to connect to Wifi
DEBUG [1008]: Stats: 87 frames/s, frame delay: 6350 us
DEBUG [2000]: Stats: 139 frames/s, frame delay: 7300 us
DEBUG [2007]: Trying to connect to Wifi
DEBUG [3002]: Trying to connect to Wifi
DEBUG [3002]: Stats: 124 frames/s, frame delay: 7500 us
DEBUG [4004]: Trying to connect to Wifi
DEBUG [4005]: Stats: 121 frames/s, frame delay: 7500 us
DEBUG [5000]: Stats: 120 frames/s, frame delay: 7500 us
DEBUG [5007]: Trying to connect to Wifi

bb-Ricardo, 2021-02-04:
Mhh, the wifi Manager doesn't even start.

bb-Ricardo, 2021-02-04:
Did you set it to 1MB SPIFFS when flashing the device?

enniob, 2021-02-04:
I did not, how can I change that?

enniob, 2021-02-04:
@bb-Ricardo I was able to figure it out. So I had to download the https://github.com/me-no-dev/arduino-esp32fs-plugin/releases/ again, and after I flash it. It worked.

bb-Ricardo, 2021-02-04:
So now it is working with ESP32 and everything? Can you post your serial console output again?

Thank you

enniob, 2021-02-04:
Sure. Here is the ouput. I removed the SSID and Password.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

DEBUG [215]: Loading config

DEBUG [227]: System Information:
DEBUG [227]: Heap: 205684
DEBUG [227]: SDK: master
DEBUG [227]: CPU Speed: 240 MHz
DEBUG [227]: Flash Size: 4194304KB
DEBUG [229]: MAC address: A4:CF:12:0B:0E:D4

*WM: [1] Free heap:        254804
*WM: [1] ESP SDK version:  v3.2.3-14-gd3e562907
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [3] setupHostname:  LEDs
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: **xxxxxx**
*WM: [3] Using Password: **xxxxx**
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] [EVENT] WIFI_REASON: 15
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresulttmp: WL_DISCONNECTED
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  LEDs - 0ED4
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             LEDs - 0ED4
*WM: [1] password:         
*WM: [1] ssid_len:         11
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [2] setting softAP Hostname: LEDs
*WM: [3] hostname: AP:  LEDs
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC completed in 2104 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] WiFi Scan completed in 2104 ms
*WM: [2] Config Portal Running, non blocking/processing 
INFO: Wi-Fi manager portal running. Connect to the Wi-Fi AP 'LEDs - 0ED4' to configure your wireless connection
DEBUG [13235]: SPIFFS contents:
DEBUG [13345]: FS File: /css/bootstrap.min.css, size: 121093
DEBUG [13347]: FS File: /css/dark-mode.css, size: 1350
DEBUG [13360]: FS File: /css/minicolors.min.css, size: 96597
DEBUG [13361]: FS File: /css/styles.css, size: 1392
DEBUG [13362]: FS File: /favicon.ico, size: 4286
DEBUG [13366]: FS File: /fonts/glyphicons.woff2, size: 18028
DEBUG [13369]: FS File: /images/atom196.png, size: 3889
DEBUG [13374]: FS File: /images/github.ico, size: 6518
DEBUG [13379]: FS File: /images/surr.png, size: 1897
DEBUG [13384]: FS File: /index.htm, size: 9706
DEBUG [13388]: FS File: /js/app.js, size: 16087
DEBUG [13396]: FS File: /js/bootstrap.min.js, size: 37045
DEBUG [13397]: FS File: /js/dark-mode-switch.min.js, size: 492
DEBUG [13413]: FS File: /js/jquery-3.1.1.min.js, size: 86709
DEBUG [13415]: FS File: /js/minicolors.min.js, size: 15463
DEBUG [13416]: FS File: /settings.htm, size: 9753

DEBUG [13434]: FS Size: 1378241KB, used: 438497KB, 31.82%

DEBUG [13435]: Enabled Features:
DEBUG [13435]: Feature: mDNS support enabled

DEBUG [13435]: Configured device type: LED strip (0)
DEBUG [13440]: NUM_LEDS: 24
DEBUG [13442]: BAND_GROUPING: 1
DEBUG [13445]: PACKET_LENGTH: 24
INFO: HTTP web server started
DEBUG [14002]: Stats: 51 frames/s, frame delay: 4550 us
DEBUG [15003]: Stats: 185 frames/s, frame delay: 7800 us
DEBUG [16007]: Stats: 116 frames/s, frame delay: 7600 us

bb-Ricardo, 2021-02-04:
this looks great, thank you.

enniob, 2021-02-04:
You are welcome.

bb-Ricardo, 2021-02-06:
merged to dev branch

closed 2021-02-06 by bb-Ricardo