NimmLor/esp8266-fastled-iot-webserver #79: Web UI not loading

blackote, 2020-11-05:
I have been able to write the code and upload to SPIFFs.  Had the NODEMCU running and worked perfect..  But occasionally run into issues where the UI does not load on a webpage every so often.  Tried re-writing to the NodeMCU both the code and sketch for SPIFFs.  

Anyone else have this problem?

Birdup, 2020-12-04:
I am also having this problem. For the first few seconds I can access the UI, then the service becomes unreachable.

blackote, 2020-12-04:
Sorry I never updated.. I noticed that if I were to get the board closer to my router I wouldn't have this issue.  So I'm thinking to add a wire to the wifi antenna to increase the range.  My router is on the 2nd floor of a 3 story townhome and my esp is on the 1st.

Birdup, 2020-12-04:
Interesting, I used another D1 Board and it resolved the issue

bb-Ricardo, 2021-01-17:
Hi,

Can you check out the newest "in-development" branch and see if the issue still persist?

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo