NimmLor/esp8266-fastled-iot-webserver #80: Mac OSX pyserial or esptool directories not found next to this upload.py tool with esp8266 2.7.x

jchilders99, 2020-11-06:
This appears to be an issue with python and the esp8266 library, not this development, but I can't find a solution anywhere.  So my advance apologies for opening an issue on your code, but I'm at a loss and hoping someone here can help.

On Mac OSX I'm unable to upload any sketches to any esp8266 boards using any version higher than esp8266 2.5.0.  The sketch compiles fine, but gives the following error:

 pyserial or esptool directories not found next to this upload.py tool.

I edited upload.py to print the expected paths for pyserial and esptool and they're correct; the folders and files exist for both pyserial and esptool.  

I was able to successfully upload the same sketch from Windows 10 with no issues and once uploaded, ArudinoOTA works without issue from the Mac.

bb-Ricardo, 2021-01-31:
Hey,

I'm doing all my development on OSX myself and never had an issue. Did you found a solution?

bb-Ricardo, 2021-03-03:
Closing this issue due to no response.

closed 2021-03-03 by bb-Ricardo