NimmLor/esp8266-fastled-iot-webserver #89: alexa configuration problem

Daniele0015, 2020-12-10:
i am creating nanoleaf and added as alexa library explained, but i am not getting the variable ENABLE_DEVICE_ALEXA so i can't program it to connect them to alexa.what should I do?

AnanasAPod, 2021-01-07:
[image]

AnanasAPod, 2021-01-07:
[image]

bb-Ricardo, 2021-01-12:
Hi, so if you enable the setting (like in the picture) it still does not work?

hugo18192, 2021-01-25:
Hi there,
Unfortunately, the device cannot be connected to Alexa. As in the description, I activated Alexa setting. After installing it on the Wemos, go to the Alexa settings page and start the search function on an Alexa. Lamp is not recognized.