NimmLor/esp8266-fastled-iot-webserver #100: non riesco a connettere l'esp con wifimanger

Mastro860, 2020-12-21:
I followed all the instructions it indicates but I can't connect to esp, it doesn't create a host network to connect to and therefore I can't open IP 192.168.4.1
You can help me thanks

spward, 2020-12-25:
Same issue. Thanks

NimmLor, 2020-12-26:
Hi, please connect to the esp8266 with the serial monitor and post the output here.

I recommend trying the following:

- Wipe the esp when flashing by changing the erase mode to `All Flash Contents`, then upload the sketch and the sketch data
[image]

- Update all libraries and the esp8266 board library to the newest version

Mastro860, 2020-12-28:
[image]

this is the serial monitor

Mastro860, 2020-12-28:
after several attempts I managed to connect it but now I can't access the site through the esp id for the configuration

Mastro860, 2020-12-28:
I tried to connect the esp to the nanoleaf and I saw that with the power connected it no longer creates a wifi network as it did before by connecting only the esp to the pc, also you can tell me how I can enable the pairing mode of the esp

bb-Ricardo, 2021-01-06:
Can you try the newest master branch. There is a fix for wifiManager included.

sarreck, 2021-01-14:
I appear to be having the same issue with the code pulled from github today. 

Everything compiles fine (including the sketch data) however once connected to the AP i cannot connect to 192.168.4.1.

I am using the OTA and MQTT options with the LED Clock config. I have tried without OTA and MQTT with the same results. 

I am using a NodeMCU 1.0 and have also tried with a D1 mini with the same results. 

All libraries are updated and i am wiping the ESP when flashing. 

Console dump is - 

```
Heap: 34880
Boot Vers: 31
CPU: 80
SDK: 2.2.1(cfd48f3)
Chip ID: 3384474
Flash ID: 1458270
Flash Size: 4194304
Vcc: 65535

SPIFFS contents:

*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  LED-CLOCK - A49A
*WM: [2] AP has anonymous access! 
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             LED-CLOCK - A49A
*WM: [1] password:         
*WM: [1] ssid_len:         16
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
mode : sta(40:f5:20:33:a4:9a) + softAP(42:f5:20:33:a4:9a)
add if0
scandone
*WM: [2] WiFi Scan completed in 2193 ms
*WM: [2] Config Portal Running, non blocking/processing 
Wi-Fi manager portal running
HTTP web server started
add 1
aid 1
station: 0e:c4:37:d1:e1:4c join, AID = 1
```

I have noticed, occasionally  (and perhaps linked to me closing my laptop and giving up for a while) That I get the following in the serial monitor:

```
station: 0e:c4:37:d1:e1:4c leave, AID = 1
rm 1
add 1
aid 1
station: 9c:b6:d0:f2:e4:bb join, AID = 1
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 29(StoreProhibitedCause):
Fatal exception 2(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception &(StoreProhibitedCause):
Fatal exception Fatal exception Fatal exception Fatal exception Fatal exception 
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
e

```

I am happy to try any suggestions to get this working. 

Thanks for the great software! 

I am really looking forward to working through this issue and getting everything working.

WarDrake, 2021-01-14:
in the NodeMCU uses different ports and the interrupts case some issues, go to my fork and try the NodeMCU branch, it's current to a few days back, see if that solves it for you, if so I'll be updating that branch to the current one on this repo pretty soon.

sarreck, 2021-01-14:
Thank you.

I have given your fork a try with the same results. Also i am having the same issue with a D1 mini.

Mastro860, 2021-01-14:
I have tried to update every library and software to the latest but I get the same results I had before

ExPeacer, 2021-01-15:
Tried with multiple boards, same issue.

bb-Ricardo, 2021-01-15:
wifiManager seems to be under heavy development these days. You need to use the development release. 

[image]

Also try to use the "in-development" branch

Mastro860, 2021-01-15:
I don't know if my problem is wifimanager as I managed to configure the wifi, the problem is that I can't open the ESP ID and alexa can't find any devices.

ExPeacer, 2021-01-16:
> wifiManager seems to be under heavy development these days. You need to use the development release.
> 
> [image]
> 
> Also try to use the "in-development" branch

Tried this, still can't connect. Can't see the notice in Serial Monitor when my device connects to the SoftAP, but there is a lots of NTP spam.

Debug log:

`DEBUG [170]: Loading config
DEBUG [171]: 
System Information:
DEBUG [171]: Heap: 36888
DEBUG [171]: Boot Vers: 31
DEBUG [171]: CPU Speed: 80 MHz
DEBUG [172]: SDK: 2.2.2-dev(38a443e)
DEBUG [175]: Chip ID: 14483326
DEBUG [177]: Flash ID: 1458208
DEBUG [180]: Flash Size: 4194304KB
DEBUG [183]: Vcc: 65535

*WM: [1] getCoreVersion():          2_7_4
*WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             34688
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  LEDs - FF7E
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             LEDs - FF7E
*WM: [1] password:         
*WM: [1] ssid_len:         11
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2183 ms
*WM: [2] Config Portal Running, non blocking/processing 
INFO: Wi-Fi manager portal running
DEBUG [3473]: SPIFFS contents:
DEBUG [3482]: FS File: /css/bootstrap.min.css, size: 121093
DEBUG [3482]: FS File: /css/dark-mode.css, size: 1104
DEBUG [3490]: FS File: /css/minicolors.min.css, size: 96597
DEBUG [3491]: FS File: /css/styles.css, size: 2152
DEBUG [3496]: FS File: /favicon.ico, size: 4286
DEBUG [3500]: FS File: /fonts/glyphicons.woff2, size: 18028
DEBUG [3505]: FS File: /images/atom196.png, size: 3889
DEBUG [3510]: FS File: /images/github.ico, size: 6518
DEBUG [3514]: FS File: /images/surr.png, size: 1897
DEBUG [3519]: FS File: /index.htm, size: 9599
DEBUG [3523]: FS File: /js/app.js, size: 16035
DEBUG [3528]: FS File: /js/bootstrap.min.js, size: 37045
DEBUG [3532]: FS File: /js/dark-mode-switch.min.js, size: 492
DEBUG [3541]: FS File: /js/jquery-3.1.1.min.js, size: 86709
DEBUG [3543]: FS File: /js/minicolors.min.js, size: 15463
DEBUG [3548]: FS File: /js/r-websocket.min.js, size: 3100
DEBUG [3553]: FS File: /settings.htm, size: 9873
DEBUG [3557]: FS File: /simple.htm, size: 3653

DEBUG [3584]: FS Size: 1953282KB, used: 446529KB, 22.86%

DEBUG [3585]: Enabled Features:
DEBUG [3585]: Feature: mDNS support enabled
DEBUG [3585]: Feature: UDP visualization support enabled

INFO: HTTP web server started
DEBUG [3645]: sending NTP packet...
DEBUG [4946]: sending NTP packet...
DEBUG [6247]: sending NTP packet...
DEBUG [7547]: sending NTP packet...
DEBUG [8848]: sending NTP packet...
DEBUG [10149]: sending NTP packet...
DEBUG [11449]: sending NTP packet...
DEBUG [12750]: sending NTP packet...
DEBUG [14050]: sending NTP packet...
DEBUG [15351]: sending NTP packet...
DEBUG [16652]: sending NTP packet...
DEBUG [17952]: sending NTP packet...
DEBUG [19253]: sending NTP packet...
`

Tried the latest release w/o WifiManager (4.3), can connect to it without any problem.

sarreck, 2021-01-16:
I have tried with yesterdays version of WifiManager with the same results. 

However if I set the "Device Type" to 0 the wifi config works fine. 

It appears the issue is when using Device Type 2 (and possibly others, i haven't tested).

bb-Ricardo, 2021-01-16:
Ahhh, good one. Will try using it with device type 2.

I can see that it is stuck in NTP Sync.

bb-Ricardo, 2021-01-16:
Hey, can you try the latest "in-development". I changed the behaviour of the NTP setup.

Set `LED_DEBUG` to `1` and keep your serial monitor open to see whats going on.

ExPeacer, 2021-01-18:
It's working right now, I can connect it multiple times, multiple devices, with Device Type 2, without any problem. Got some new ones, but this one is working.

sarreck, 2021-01-19:
Thank you. 

This has resolved the issue for me.

bb-Ricardo, 2021-01-19:
Thank you for testing it.

I would like to maje some more changes to the NTP code. Do I understand it correctly that if day light saving changes, you have to reflash your device?

bb-Ricardo, 2021-02-02:
Closing this issue as resolved.

closed 2021-02-02 by bb-Ricardo