NimmLor/esp8266-fastled-iot-webserver #106: Wemos D1 hang when too many web request with WifiManager

metalheadz, 2021-01-02:
Hi!

I encountered a problem where too many web requests in a short period of time would hang my Wemos while using the UI or REST endpoints.

When the Wemos was not responding, Ive seen that the AP mode was active, which was not normal.

I just removed WifiManager and made a direct wifi connection and all problems are gone. 

I dont know if this problem is related to my particular setup or a bug, but I think WifiManager add an unnecessary overhead in the main loop while most of people dont need AP mode as these device are permanently attached at the same place.

Would be nice to get the option to avoid using manager and get a direct connection. Looking at the code, I guess it was like that before with the secrets.h file but removed to implement WifiManager.

Thanks!

bb-Ricardo, 2021-01-12:
Hi, wanted to to add a performance counter to check the loop cycles per second.

This is something to look into. The wifiManger loop could be disabled as soon as the wifi is setup and running.

bb-Ricardo, 2021-01-15:
Hi, can you check the latest "in-development" branch. I added debugging and performance measuring details.

This should help you a bit better to find the bottleneck.

How many LEDs are you addressing?

bb-Ricardo, 2021-01-31:
Closing this issue due to no response. If you still having troubles consider opening a new issue.

closed 2021-01-31 by bb-Ricardo