NimmLor/esp8266-fastled-iot-webserver #107: MQTT Discovery

arnaud-42, 2021-01-04:
Hello,

I can't make MQTT Discovery work in Home Assistant.
My Device is DEVICE_TYPE 4

Here HA log error : 

> Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'name': 'Nanoleafs', 'device': {'identifiers': 'E0:98:06:0D:9D:05', 'manufacturer': 'Surrbradl08', 'model': '0.4', 'name': 'Nanoleafs'}, 'state_topic': 'homeassistant/light/nanoleafs', 'command_topic': 'homeassistant/light/nanoleafs/set', 'brightness': True, 'rgb': True, 'effect': True, 'unique_id': 'E0:98:06:0D:9D:05', 'schema': 'json', 'effect_list': ['Pride', 'Color Waves', 'Horizontal Rainbow', 'Solid Rainbow', 'Confetti', 'Sinelon', 'Beat', 'Juggle', 'Fire', 'Water', 'Strobe', 'Rainbow Strobe', 'Smooth Rainbow Strobe', 'Rainbow Roll', 'Rainbow Beat', 'Palette Fades', 'Rainbow Chase', 'Rainbow Dots', 'Rainbow Fades', 'Police Lights', 'Glitter', 'Snow Flakes', 'Lightning', 'Rainbow Twinkles', 'Snow Twinkles', 'Cloud Twinkles', 'Incandescent Twinkles', 'Retro C9 Twinkles', 'Red & White Twinkles', 'Blue & White Twinkles', 'Red, Green & White Twinkles', 'Fairy Light Twinkles', 'Snow 2 Twinkles', 'Holly Twinkles', 'Ice Twinkles', 'Party Twinkles', 'Forest Twinkles', 'Lava Twinkles', 'Fire Twinkles', 'Cloud 2 Twinkles', 'Ocean Twinkles', 'Solid Volume Visualizer', 'Static Rainbow Volume Visualizer', 'Flowing Rainbow Volume Visualizer', 'Tri-Color Volume Visualizer', 'Wave Visualizer', 'Center Visualizer', 'Solid-Color Pair Bullet Visualizer', 'Solid-Color Complementary Bullet Visualizer', 'Blue/Purple Bullet Visualizer', 'Beat-Bullet Visualization', 'Bass Ring Visualizer', 'Kick Ring Visualizer', None, None], 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 55, in async_discover config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 340, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 336, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 32, in validate_mqtt_light return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: string **value is None @ data['effect_list'][53]** 

When i look in my MQTT broker, I've got : 
If think the 2x Null are the problem ... but i can't find why in the code.

> Message 2 received on homeassistant/light/nanoleafs/config at 8:44 PM:
> 
> {
>     "~": "homeassistant/light/nanoleafs",
>     "name": "Nanoleafs",
>     "dev": {
>         "ids": "E0:98:06:0D:9D:05",
>         "mf": "Surrbradl08",
>         "mdl": "0.4",
>         "name": "Nanoleafs"
>     },
>     "stat_t": "~",
>     "cmd_t": "~/set",
>     "brightness": true,
>     "rgb": true,
>     "effect": true,
>     "uniq_id": "E0:98:06:0D:9D:05",
>     "schema": "json",
>     "effect_list": [
>         "Pride",
>         "Color Waves",
>         "Horizontal Rainbow",
>         "Solid Rainbow",
>         "Confetti",
>         "Sinelon",
>         "Beat",
>         "Juggle",
>         "Fire",
>         "Water",
>         "Strobe",
>         "Rainbow Strobe",
>         "Smooth Rainbow Strobe",
>         "Rainbow Roll",
>         "Rainbow Beat",
>         "Palette Fades",
>         "Rainbow Chase",
>         "Rainbow Dots",
>         "Rainbow Fades",
>         "Police Lights",
>         "Glitter",
>         "Snow Flakes",
>         "Lightning",
>         "Rainbow Twinkles",
>         "Snow Twinkles",
>         "Cloud Twinkles",
>         "Incandescent Twinkles",
>         "Retro C9 Twinkles",
>         "Red & White Twinkles",
>         "Blue & White Twinkles",
>         "Red, Green & White Twinkles",
>         "Fairy Light Twinkles",
>         "Snow 2 Twinkles",
>         "Holly Twinkles",
>         "Ice Twinkles",
>         "Party Twinkles",
>         "Forest Twinkles",
>         "Lava Twinkles",
>         "Fire Twinkles",
>         "Cloud 2 Twinkles",
>         "Ocean Twinkles",
>         "Solid Volume Visualizer",
>         "Static Rainbow Volume Visualizer",
>         "Flowing Rainbow Volume Visualizer",
>         "Tri-Color Volume Visualizer",
>         "Wave Visualizer",
>         "Center Visualizer",
>         "Solid-Color Pair Bullet Visualizer",
>         "Solid-Color Complementary Bullet Visualizer",
>         "Blue/Purple Bullet Visualizer",
>         "Beat-Bullet Visualization",
>         "Bass Ring Visualizer",
>         "Kick Ring Visualizer",
**>         null,
>         null**
>     ]
> }

arnaud-42, 2021-01-04:
So i try a lot a thing.
It work when i remove this effects :

    //{ NanoleafWaves, "Nanoleaf Wave Visualizer"},
    //{ NanoleafBand, "Nanoleaf Rainbow Band Visualizer"},
    //{ NanoleafSingleBand, "Nanoleaf Solid Color Band Visualizer"},

I can't find why when they are includes i've got null value send to mqtt ...

WarDrake, 2021-01-04:
Yes, this is an error on the transfer size when you enable visualizations, i have a fix that i'll push in a little bit.

closed 2021-01-10 by arnaud-42