NimmLor/esp8266-fastled-iot-webserver #114: 4 Second Freezes

CrimsonAC, 2021-01-07:
Hello and first of all, thanks for the great software and effort that's put into it.

I have a strange problem since the new 4.4 release.
The Wemos D1 Mini just seems to freeze for 4 seconds, then it works for 4 seconds, then it freezes again. 
This goes on and on, always in a approximately 4 seconds interval.
I didn't change anything in my setup between 4.3 and 4.4.

My setup:
12V Power Supply to BuckConverter -> 5V to Wemos D1 Mini.
12V Power Supply to W2811 LED-Stripe.
Wemos D1 Mini Data Pin to LED-Stripe.

I tried a second D1 Mini with the same result.
The web interface seems to lag during that phases, too.
For example, when switching between on and off, during those freeze times, the Stripe does not change and the web interface says "please wait" in the status line.

Going back to 4.3 and everything works again.

scr1ptDev, 2021-01-07:
Same problem. Start the program correctly, after a few seconds, it crashes. On different occasions, it crashes every 4 seconds, on other occasions it does not run the animations.
Indicate that it create an access point called "ESP-13E3F1" when in reality the hostname is "Ledlamp". The webserver is very, very slow.
When I mounted it yesterday, it worked perfect, it was to install IoT-Audio-Visualization-Center and stop working. I have flashed Wemos again, and nothing. Sorry english :)

bb-Ricardo, 2021-01-09:
Can you add the serial output here? It looks like it is restarting in a loop. Are you using "Autoplay"?

bb-Ricardo, 2021-01-15:
can you test the latest "in-development" branch and check the Serial Monitor?

CrimsonAC, 2021-01-15:
Oh sorry, I didn't notice the reply in which you wanted to see the serial output.
I just tested the mentioned "in-development" branch. Everything seems to work fine. 
Here is a part of the serial output. I changed the patterns once or twice and did a reboot while the monitor was active (i blanked my wifi credentials).

14:33:39.086 -> DEBUG [42002]: Stats: 120 frames/s, frame delay: 7950 us
14:33:40.066 -> DEBUG [43004]: Stats: 120 frames/s, frame delay: 7950 us
14:33:41.084 -> DEBUG [44006]: Stats: 120 frames/s, frame delay: 7950 us
14:33:42.083 -> DEBUG [45007]: Stats: 120 frames/s, frame delay: 7950 us
14:33:43.071 -> DEBUG [46000]: Stats: 119 frames/s, frame delay: 7950 us
14:33:44.052 -> DEBUG [47000]: Stats: 120 frames/s, frame delay: 7950 us
14:33:45.071 -> DEBUG [48003]: Stats: 120 frames/s, frame delay: 7950 us
14:33:46.061 -> DEBUG [49005]: Stats: 120 frames/s, frame delay: 7950 us
14:33:47.083 -> DEBUG [50006]: Stats: 120 frames/s, frame delay: 7950 us
14:33:48.074 -> DEBUG [51000]: Stats: 119 frames/s, frame delay: 7950 us
14:33:49.059 -> DEBUG [52001]: Heap: 27536
14:33:49.059 -> DEBUG [52002]: Stats: 120 frames/s, frame delay: 7950 us
14:33:50.076 -> DEBUG [53004]: Stats: 120 frames/s, frame delay: 7950 us
14:33:50.244 -> DEBUG [53191]: Setting: pattern: Pride
14:33:51.088 -> DEBUG [54006]: Stats: 118 frames/s, frame delay: 7850 us
14:33:52.075 -> DEBUG [55003]: Stats: 121 frames/s, frame delay: 7850 us
14:33:53.061 -> DEBUG [56001]: Stats: 121 frames/s, frame delay: 7850 us
14:33:54.074 -> DEBUG [57004]: Stats: 122 frames/s, frame delay: 7950 us
14:33:55.090 -> DEBUG [58005]: Stats: 120 frames/s, frame delay: 7950 us
14:33:56.075 -> DEBUG [59004]: Stats: 120 frames/s, frame delay: 7950 us
14:33:57.060 -> DEBUG [60003]: Stats: 120 frames/s, frame delay: 7950 us
14:33:58.078 -> DEBUG [61005]: Stats: 120 frames/s, frame delay: 7950 us
14:33:59.083 -> DEBUG [62004]: Heap: 27368
14:33:59.083 -> DEBUG [62005]: Stats: 120 frames/s, frame delay: 7950 us
14:34:00.060 -> DEBUG [63005]: Stats: 120 frames/s, frame delay: 7950 us
14:34:00.265 -> DEBUG [63198]: Saving Config
14:34:00.467 -> DEBUG [63399]: Comitting config
14:34:01.076 -> DEBUG [64000]: Stats: 113 frames/s, frame delay: 7600 us
14:34:02.078 -> DEBUG [65005]: Stats: 126 frames/s, frame delay: 7900 us
14:34:03.069 -> DEBUG [66005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:04.089 -> DEBUG [67006]: Stats: 121 frames/s, frame delay: 7900 us
14:34:05.074 -> DEBUG [68008]: Stats: 121 frames/s, frame delay: 7900 us
14:34:06.086 -> DEBUG [69003]: Stats: 120 frames/s, frame delay: 7900 us
14:34:07.076 -> DEBUG [70005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:08.074 -> DEBUG [71007]: Stats: 121 frames/s, frame delay: 7900 us
14:34:09.062 -> DEBUG [72000]: Heap: 27368
14:34:09.062 -> DEBUG [72001]: Stats: 120 frames/s, frame delay: 7900 us
14:34:10.081 -> DEBUG [73002]: Stats: 121 frames/s, frame delay: 7900 us
14:34:11.061 -> DEBUG [74003]: Stats: 121 frames/s, frame delay: 7900 us
14:34:12.060 -> DEBUG [75005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:13.077 -> DEBUG [76008]: Stats: 121 frames/s, frame delay: 7900 us
14:34:14.054 -> DEBUG [77000]: Stats: 120 frames/s, frame delay: 7900 us
14:34:15.083 -> DEBUG [78002]: Stats: 121 frames/s, frame delay: 7900 us
14:34:16.064 -> DEBUG [79003]: Stats: 121 frames/s, frame delay: 7900 us
14:34:17.073 -> DEBUG [80005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:18.064 -> DEBUG [81008]: Stats: 121 frames/s, frame delay: 7900 us
14:34:19.075 -> DEBUG [82000]: Heap: 27368
14:34:19.075 -> DEBUG [82001]: Stats: 120 frames/s, frame delay: 7900 us
14:34:20.087 -> DEBUG [83002]: Stats: 121 frames/s, frame delay: 7900 us
14:34:21.066 -> DEBUG [84003]: Stats: 121 frames/s, frame delay: 7900 us
14:34:22.089 -> DEBUG [85004]: Stats: 121 frames/s, frame delay: 7900 us
14:34:23.072 -> DEBUG [86005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:24.081 -> DEBUG [87005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:25.067 -> DEBUG [88007]: Stats: 121 frames/s, frame delay: 7900 us
14:34:26.062 -> DEBUG [89004]: Stats: 110 frames/s, frame delay: 7400 us
14:34:27.056 -> DEBUG [90000]: Stats: 103 frames/s, frame delay: 6550 us
14:34:28.073 -> DEBUG [91003]: Stats: 145 frames/s, frame delay: 7800 us
14:34:29.062 -> DEBUG [92000]: Stats: 122 frames/s, frame delay: 7900 us
14:34:29.062 -> DEBUG [92008]: Heap: 27480
14:34:30.088 -> DEBUG [93005]: Stats: 100 frames/s, frame delay: 6900 us
14:34:31.079 -> DEBUG [94000]: Stats: 137 frames/s, frame delay: 7750 us
14:34:32.073 -> DEBUG [95000]: Stats: 123 frames/s, frame delay: 7900 us
14:34:33.067 -> DEBUG [96000]: Stats: 121 frames/s, frame delay: 7900 us
14:34:34.061 -> DEBUG [97003]: Stats: 121 frames/s, frame delay: 7900 us
14:34:35.084 -> DEBUG [98007]: Stats: 121 frames/s, frame delay: 7900 us
14:34:36.139 -> 
[... 64 lines cut ...]
14:34:39.425 -> 
14:34:39.425 -> DEBUG [3254]: Enabled Features:
14:34:39.425 -> DEBUG [3257]: Feature: mDNS support enabled
14:34:39.425 -> DEBUG [3261]: Feature: OTA support enabled
14:34:39.425 -> DEBUG [3265]: Feature: MQTT support enabled
14:34:39.425 -> DEBUG [3268]: Feature: UDP visualization support enabled
14:34:39.460 -> 
14:34:39.460 -> INFO: HTTP web server started
14:34:40.347 -> INFO: WiFi Connected! Open http://192.168.2.57 in your browser
14:34:40.347 -> INFO: mDNS responder started. Try to open http://newLEDs.local in your browser
[... cut ...]
14:34:47.501 -> DEBUG [11002]: Stats: 120 frames/s, frame delay: 7900 us
14:34:48.169 -> DEBUG [12000]: Stats: 120 frames/s, frame delay: 7900 us
14:34:49.186 -> DEBUG [13005]: Heap: 27824
14:34:49.186 -> DEBUG [13006]: Stats: 121 frames/s, frame delay: 7900 us
14:34:50.172 -> DEBUG [14003]: Stats: 120 frames/s, frame delay: 7900 us
14:34:51.157 -> DEBUG [15005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:52.178 -> DEBUG [16005]: Stats: 121 frames/s, frame delay: 7900 us
14:34:53.164 -> DEBUG [17000]: Stats: 120 frames/s, frame delay: 7900 us

bb-Ricardo, 2021-01-15:
great.

@scr1ptDev: can you test it as well?

bb-Ricardo, 2021-02-02:
Closing this issue as resolved. Thank you for testing

closed 2021-02-02 by bb-Ricardo