NimmLor/esp8266-fastled-iot-webserver #115: Few documentation issues

pat-bohn, 2021-01-09:
Hi, I am trying to follow the Tutorial here: https://github.com/NimmLor/esp8266-fastled-iot-webserver/blob/master/Software_Installation.md 

I have a D1 Mini and thus set it up like this:
[image]

The installed versions are:
Windows 10
Arduino 1.8.13

esp8266 2.7.0

FastLED 3.3.2
WiFiManager 2.0.3-alpha 

PubSubClient 2.7.0
ArduinoJson 6.15.1

I came across two/three issues:
- when activating mqtt support, variables (mqttuser and password) in the arduino.ino scatch below the "config end" have to be changed or a compilation error will occur. 
- The installation path for the ESP8266 data upload tool (under Windows) is /Arduino/tools/ESP8266FS/tool/esp8266fs.jar; not /Arduino/tools/esp8266fs.jar
- OTA appears to not work after first flash of the esp8266-fastled-iot-webserver.ino sketch (network COM port disappears)

bb-Ricardo, 2021-01-14:
Did you try the latest master? It should compile without changing the mqtt settings in the scetch file.

bb-Ricardo, 2021-02-02:
Hi, did you have time to try out the latest "in-development" branch?

bb-Ricardo, 2021-02-09:
Closing this issue for now. A lot has been updated. Please check the 'in-development' branch and see if this fixes it for you

closed 2021-02-09 by bb-Ricardo