NimmLor/esp8266-fastled-iot-webserver #117: Request: Homey integration

MaikelKelder, 2021-01-11:
Homey is a popular Demotica system in Europe.

Integration is described here:
https://github.com/athombv/homey-arduino-library/blob/master/docs/usage.md

I am experimenting, but as a total new guy to it, I probably not gonna make it work 100%.
If someone could help, much appreciated!

Gupzilla, 2021-01-11:
Also interested in this feature, as this will make the solution more smart.

bb-Ricardo, 2021-01-12:
Hi,

From the looks of it the basic integration would be pretty straight forward.

BUT: I personally don't have a homey device to test the integrations and actions.

Do you have an idea of how I would be able to test it without an homey device/pod?

MaikelKelder, 2021-01-12:
The only thing I can offer is that I will try out any code you want to test?
Homey doesnt have a virtual setup or something, and the device is too costly to just buy to test ;-)
What can I do to make this easier for you?

MaikelKelder, 2021-01-12:
You may need... (or I do)
Homey MAC: 2c:6b:7d:14:82:96
Homey IP: 192.168.178.216

MaikelKelder, 2021-01-12:
I was able to register the Arduino (Wemos) to Homey by adding 3 lines lines:
(As mentioned here: https://github.com/athombv/homey-arduino-library/blob/master/docs/usage.md)
Took out the extra lines, that dont seem to be needed (wifi SSID, username and pass)

const char* name = "NanoLeafs";

void setup() {
  Homey.begin(name);
}

void loop() {
  Homey.loop();
}

With these added and builded, I can register the Arduino to the App in Homey, yet, cant control anything of course.
This is where my knowledge ends a bit, I am afraid... I dont know how to control the lights..

bb-Ricardo, 2021-01-12:
Jup, this part is easy 😀.

I meant the actions to control the device is difficult to test.

Can you point me to a doku where all the actions are described? We could start with a simple switch on/off command.

MaikelKelder, 2021-01-12:
I am looking for a more constructial guide for commands. but hope this page helps?
https://apps-sdk-v2.developer.athom.com/tutorial-App%20Development%20Guidelines.html
(Athom is the brand of Homey)

MaikelKelder, 2021-01-12:
I Found also an example INO project on the Homey Arduino Github page:
https://github.com/athombv/homey-arduino-library/tree/master/examples/Example%20projects/Community%20projects/LedStripDriver_OpenSmart

bb-Ricardo, 2021-01-12:
This seems to be a good source. Thank you.

On/off and brightness seems pretty forward. But I'm not sure how patterns should be controlled.

We could just simply add these two. Even though it's possible to change the color. This would make it necessary to switch to the solid color pattern.

What do you think?

MaikelKelder, 2021-01-12:
Sounds Great, ON/OFF would already be great since it remembers it last setting.
Lets worry/experiment on the rest later :-p

Thanks for this, it greatly appreciated!

bb-Ricardo, 2021-01-12:
I should be able to whip something up tonight.

MaikelKelder, 2021-01-12:
*Speechless* ;-)

bb-Ricardo, 2021-01-12:
Just did a little change. See: https://github.com/bb-Ricardo/esp8266-fastled-iot-webserver/commit/8e44930e7ccacbe00b5bb1a278cf1a7df3ffeab0

It compiles but I can't test it. Included On/Off and brightness.

Can you try it out?

MaikelKelder, 2021-01-12:
I will! let you know ASAP!

MaikelKelder, 2021-01-12:
Getting an error on compile, but I also get that on the original ino, now placed on in master. (6 days old)
Not getting it on the older version btw.

Error is in screenshot: "cfg was not declared
[image]

bb-Ricardo, 2021-01-13:
I recently changed the config management. There should be a config.h file which should be included in line 374 of that scatch. Is this file missing/empty? Or may check the end if line.

Thank you.

MaikelKelder, 2021-01-13:
Yes, now it builds and uploads.
Homey seems to see it as a switch/dimmable device and lets me switch.
Website doesnt seem to load anymore 100%, see screenshot, tried reuploading Sketch Data, but didnt change anything.
Its 1 AM now, going to sleep. will test with leds tomorrow.
Thanks so far!!!

[image]

bb-Ricardo, 2021-01-13:
Great, step by step. Did you figure out the config.h issue and why it wouldn't compile the first time?

Will add a counter later on so you can see how many loop cycles per seconds are processed. This way you can tell on the serial console if homey has a big performance impact.

bb-Ricardo, 2021-01-13:
Just pushed another change which will display a loop/s counter in your serial console. Make sure NOT to use "sinelon" als pattern, it is currently broken.

MaikelKelder, 2021-01-13:
> Great, step by step. Did you figure out the config.h issue and why it wouldn't compile the first time?
> 
> Will add a counter later on so you can see how many loop cycles per seconds are processed. This way you can tell on the serial console if homey has a big performance impact.

I used the new master with config.h as you mentioned, my build was still without. With the config.h it builded.
I will try your newest and see if webpage also shows again..... I will do a full scrub of the WEmos.

MaikelKelder, 2021-01-13:
Sorry, had a challenge with deleting an old device in Homey from earlier registration, as long as that existed, it didnt let me register a new one, same device, but somehow it had a conflict.
Worked that out, and whatayaknow... IT WORKS!

ON/OFF even the DIMM works! Can even put in in a flow and works!!

bb-Ricardo, 2021-01-13:
This seems a good value. great. I could even try to add a hue setting. Are you around for some testing any longer?

MaikelKelder, 2021-01-13:
Absolutely up for more testing

bb-Ricardo, 2021-01-13:
Great, latest commit adds support for color setting, can you test: https://github.com/NimmLor/esp8266-fastled-iot-webserver/commit/f7feec10cf6a3fe1a3a649562c2395b34939c745

MaikelKelder, 2021-01-13:
I do get a HUE control in Homey, but it doesnt change the color of the LEDS, they change to default white when selecting the HUE and stay white.

bb-Ricardo, 2021-01-13:
try changing Line 1563:
`rainbowSolid();`
with
`fill_solid(leds, NUM_LEDS, CHSV(gHue, 128, 255));` or `fill_solid(leds, NUM_LEDS, CHSV(gHue, 0, 255));`

I'm just about adding saturation.

MaikelKelder, 2021-01-13:
Getting this error on compile:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Maikel\Downloads\__MASTER\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:0:

C:\Users\Maikel\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^

In file included from C:\Users\Maikel\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\Maikel\Downloads\__MASTER\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:20:

C:\Users\Maikel\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\Maikel\Downloads\__MASTER\esp8266-fastled-iot-webserver-master\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino: In function 'void homeyLightHue()':

esp8266-fastled-iot-webserver:1563:52: error: expected primary-expression before 'or' token

  fill_solid(leds, NUM_LEDS, CHSV(gHue, 128, 255)); or fill_solid(leds, NUM_LEDS, CHSV(gHue, 0, 255));

                                                    ^

exit status 1

expected primary-expression before 'or' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bb-Ricardo, 2021-01-13:
:D

This was meant as one of them. Try just `fill_solid(leds, NUM_LEDS, CHSV(gHue, 128, 255)); `

MaikelKelder, 2021-01-13:
And just now I realised that :-p

bb-Ricardo, 2021-01-13:
can you change it back and add a `delay(5000);` after that line

MaikelKelder, 2021-01-13:
Back to Rainbowsolid?

bb-Ricardo, 2021-01-13:
Well the just to this:
```
fill_solid(leds, NUM_LEDS, CHSV(gHue, 255, 255));
delay(5000);
```

MaikelKelder, 2021-01-13:
No change in color, but do get some LAG now in the interface of homey and sometimes timeout errors.

bb-Ricardo, 2021-01-13:
Na, this was a silly idea. But i know where the issue is. Just trying to fix it properly.

MaikelKelder, 2021-01-13:
I am hitting the sack its getting late, but again, amazing stuff, so happy this is possible and working so well.

bb-Ricardo, 2021-01-13:
fair enough. I just pushed another commit. Let's see if this changes anything

MaikelKelder, 2021-01-13:
No color change still, same result it seems.

bb-Ricardo, 2021-01-13:
sorry, had another bug

pushed a new commit

MaikelKelder, 2021-01-14:
I wish I could show you the result of your work!! It WORKS!!!
I think you created in 3 days a great integration!!
Do you think its possible to integrate the Pattern selection, or is that not possible because of Homey limitations?

bb-Ricardo, 2021-01-14:
Awesome 👏🏼

> Do you think its possible to integrate the Pattern selection, or is that not possible because of Homey limitations?

If Homey has an API options to select a pattern, it should also be quite easy to add.

Can you go gunt for the docu again, if you can find anything? We are using the device class "light".

MaikelKelder, 2021-01-14:
absolutely, will look for it!

MaikelKelder, 2021-01-14:
https://apps-sdk-v2.developer.athom.com/index.html

I dont know if its of any use, this is the UI trigger:
[image]

There is API information in this section, just dont know if its receiving or sending....
https://apps-sdk-v2.developer.athom.com/Api.html

Sorry, try to find the right information for you, just dont know enough about this stuff ;-)

MaikelKelder, 2021-01-14:
Are Webhooks of any use? also mentioned in the same developer site.....

If none of these help, I will look somewhere else, let me know...

bb-Ricardo, 2021-01-14:
Found this here: https://developer.athom.com/docs/apps-reference

But can't find anything matching

bb-Ricardo, 2021-01-14:
Maybe "Next" and "Previous" seem fitting.

MaikelKelder, 2021-01-14:
Is light_mode an option? I dont know what is the output in the UI for it... 
Seems like the picker is not available for light.

bb-Ricardo, 2021-01-14:
Just pushed another commit. Can you test if you can change the patterns now?

MaikelKelder, 2021-01-14:
Getting an error:
[image]

bb-Ricardo, 2021-01-14:
Are you sure your build is clean? There have only been added a few lines and it compiled cleanly for me.

MaikelKelder, 2021-01-14:
no idea what happend, but yeah it was me....
I compiled normal now and yes, I have a Music option now BACK and Forward, Nanoleafs respond, so it works.
No idea what I am selecting though... ;-)

If you can get this MEDIA option in there, would a "picker"also be possible from the "thermostat" capability?
Yet this list need to be filled of course, no idea how that would take place....

bb-Ricardo, 2021-01-15:
These buttons change the pattern forth and back. These options show up as mesic options? They are defined als "speaker_next" and "speaker_prev".

Here: https://developer.athom.com/docs/apps-reference under Capabilities you can see the possible options. And "enum" is what you want. If you klick on the three dots on the right side you get the definition code. And under "enum" you can only select the predefined options.

MaikelKelder, 2021-01-15:
Yes they work ! They just look strange in the homey ui. :-)

bb-Ricardo, 2021-01-16:
Can we leave it like that? Then I would consider this solved and merge it to the development branch.

Or should we remove the pattern changing so it doesn't look to confusing.

MaikelKelder, 2021-01-17:
Absolutely, thanks for your time and effort in this!! really appreciated!

bb-Ricardo, 2021-01-17:
Thank you. Merged it to the dev branch.

commit 1d7a5936 (2021-01-17): adds Homey hub integration

The MACRO DEVICE_TYPE had to be renamed LED_DEVICE_TYPE as
DEVICE_TYPE is used in "Homey.h" as well.

fixes: #117
closed 2021-01-17 by bb-Ricardo