NimmLor/esp8266-fastled-iot-webserver #118: Compiling Issues and cant connect to the webui

m50008, 2021-01-11:
Hi, i have two main issues.
The first one, I can not compile the file if the Espalexa lib is not installed, even if the `#define ENABLE_ALEXA_SUPPORT`  is commented out. If the Espalexa lib is installed and `#define ENABLE_ALEXA_SUPPORT` is commented out, I get the following error:
So when defining AlexaSupport it compiles. 
But this leades to the secound issue. When uploading, the connecting with the esp workes and i am able to edit my wifi credentials and the esp connects to my wifi.
But when I try to connect to the ESP via the Browser, it just loades a page with "Not found". 
What should i do now?
Thanks in advance!

I build the code for the 3D-Printed Desk Lamp, so apart from changing the `DEVICE_TYPE` to 3 and enabling `#define ENABLE_ALEXA_SUPPORT` to compile the sketch i did not change anything to the code.
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_lite.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 mini Lite
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mklittlefs 1.203.200522 (2.3)
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting esp8266-fastled-iot-webserver.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.4.0
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <WiFiManager> 2.0.4-beta
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Espalexa> 0.0.0-alpha+sha.09981160b5
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\d1_mini_lite\src\esp8266-fastled-iot-webserver.ino.cpp.o
In file included from C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:473:0:
src\TwinkleFOX.h:96:1: warning: multi-line comment [-Wcomment]
 //     / \
 ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:4376:8: warning: extra tokens at end of #endif 
directive [enabled by default]
                     leds[pos - 1 + RING_LENGTH] = CHSV(145, 255, b);
        ^
In file included from C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:20:0:
.pio\libdeps\d1_mini_lite\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
In file included from .pio\libdeps\d1_mini_lite\FastLED\src/FastLED.h:65:0,
                 from C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:20:
.pio\libdeps\d1_mini_lite\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
[... 144 lines cut ...]
         ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino: In function 'void peakVisualizer(CHSV, bool)':
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:3127:17: warning: statement has no effect [-Wunused-value]
     int v = 3;
                 ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino: In function 'void ambilight()':
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:3969:13: warning: unused variable 'ledCount' [-Wunused-variable]
     static bool is_initialized = false;
             ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:4001:36: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         is_initialized = true;
                                    ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:3962:56: warning: unused variable 'i' [-Wunused-variable]
         }
                                                        ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino: In function 'int getCounterHue(int)':
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:3417:1: warning: control reaches end of non-void function [-Wreturn-type]
     if (hSwitch >= grpsz) {
 ^
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino: In function 'int getPairHue(int)':
C:/Users/Alex/Documents/PlatformIO/Projects/210111-195515-d1_mini_lite/src/esp8266-fastled-iot-webserver.ino:3422:1: warning: control reaches end of non-void function [-Wreturn-type]
     lastVals[i_pos] = currentVolume;
 ^
*** [.pio\build\d1_mini_lite\src\esp8266-fastled-iot-webserver.ino.cpp.o] Error 1
```

WarDrake, 2021-01-11:
The alexa code looks properly setup to me, are you using the latest version of the master branch?
as for the ui, you didn't mention uploading the data to the esp, so I assume you're missing that step.

bb-Ricardo, 2021-01-12:
Hi, 

1. can you provide me your platform io ini? Then I can try to get rid of the compile errors.

2. In the installation instructions it is described how to upload the webinterface files to the ESP. But this involves the Ardione IDE. There are other possibilities to do that but they are currently not implemented. Same here, I would try to fix that.

Platform IO support would be great. 😀

tommyseus, 2021-01-18:
Thanks for the hint with the alexa constant. I have also issues compiling issues when alexa is not enabled.

My platformio.ini with the `data_dir` configuration. Data_dir points to the data folder with the assets.
```plain
; platformio.ini
[platformio]
data_dir = src/data

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps = 
	fastled/FastLED@^3.4.0
	tzapu/WiFiManager@^2.0.4-beta
	aircoookie/Espalexa@0.0.0-alpha+sha.09981160b5
monitor_speed = 115200
```

To upload the files to the filesystem you must run the task.
```plain
Main Menu --> Terminal --> Run Task --> Platformio: Upload Filesystem Image (d1_mini)
```

bb-Ricardo, 2021-01-18:
Thank you for the input. I will test the setup tomorrow.

bb-Ricardo, 2021-01-19:
I imported the project into platform.io. 
Renamed `src/esp8266-fastled-iot-webserver.ino` to `src/esp8266-fastled-iot-webserver.cpp`

Then added following file to source:
`src/esp8266-fastled-iot-webserver.h`
```c
void sendInt(int);
void broadcastInt(String, int);
void ShiftLeds(int);
void SendLeds(CHSV, int);
void ShiftLedsCenter(int);
void SendLedsCenter(CHSV);
void pride();
void colorWaves();
void rainbow();
void rainbowSolid();
void confetti();
void sinelon();
void bpm();
void juggle();
void fire();
void water();
void strobe();
void rainbow_strobe();
void smooth_rainbow_strobe();
void rainbowRoll();
void rainbowBeat();
void randomPaletteFades();
void rainbowChase();
void randomDots();
void randomFades();
void policeLights();
void glitter();
void snowFlakes();
void lightning();
void vuMeterSolid();
void vuMeterStaticRainbow();
void vuMeterRainbow();
void vuMeterTriColor();
void RefreshingVisualizer();
void CentralVisualizer();
void SolidColorDualTone();
void SolidColorComplementary();
void BluePurpleBullets();
void BulletVisualizer();
void RainbowBassRings();
void RainbowKickRings();
void RainbowBandVisualizer();
void SingleColorBandVisualizer();
void showSolidColor();

void setPattern(int);
void broadcastString(String, String);
void loadConfig();
void setPower(int);
void setSpeed(int);
void sendString(String);
void setPatternName(String);
void setPalette(int);
void setPaletteName(String);
void setBrightness(int);
void setAutoplay(int);
void setAutoplayDuration(int);
void adjustPattern(int);
bool isValidHostname(char*, long);
void addGlitter(int);
void heatMap(CRGBPalette16, int);
void colorwaves(CRGB*, uint16_t, CRGBPalette16&);
bool FadeUp(CRGB, int, int, int, int, int);
CHSV getVisualizerBulletValue(double);
void SendTrailingLeds(CHSV, int);
void kickRingVisualizer(CHSV, int);
void bassRingVisualizer(CHSV, int);
void paintRing(CHSV, int, int, int);
CHSV getFadedColor(CHSV, int, int);
```

then included it to `src/esp8266-fastled-iot-webserver.cpp` after `#include "Field.h"`
```c
#include "esp8266-fastled-iot-webserver.h"
```

Used following `platformio.ini`
```ini
[platformio]
data_dir = src/data

[env:d1]
platform = espressif8266
board = d1
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
lib_deps = 
	fastled/FastLED@^3.4.0
	tzapu/WiFiManager@^2.0.4-beta
monitor_speed = 115200
upload_speed = 460800

build_flags =
  -Wno-switch
  -Wno-deprecated-declarations
  -Wno-write-strings
  -Wno-unused-variable
  -Wno-unused-value
  -Wno-sign-compare
  -Wno-unused-but-set-variable
  -Wno-return-type
  -Wno-sequence-point
  -Wno-narrowing
```

It seems to compile without any problems but the linker "ld" has issues to link everything together.

```sh
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1/src/esp8266-fastled-iot-webserver.cpp.o:(.text._Z5waterv+0x8): undefined reference to `heatMap(CRGBPalette16, int)'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1/src/esp8266-fastled-iot-webserver.cpp.o: in function `water()':
esp8266-fastled-iot-webserver.cpp:(.text._Z5waterv+0x22): undefined reference to `heatMap(CRGBPalette16, int)'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1/src/esp8266-fastled-iot-webserver.cpp.o: in function `fire()':
esp8266-fastled-iot-webserver.cpp:(.text._Z4firev+0x18): undefined reference to `heatMap(CRGBPalette16, int)'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1/src/esp8266-fastled-iot-webserver.cpp.o:(.text._Z13setSolidColorhhhb+0x4): undefined reference to `setPattern(int)'
/Users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/d1/src/esp8266-fastled-iot-webserver.cpp.o:(.text._Z13setSolidColorhhhb+0x4f): undefined reference to `setPattern(int)'
```

I'm a bit lost here, can someone help?