NimmLor/esp8266-fastled-iot-webserver #119: Nanoleaf pattern don't work

HexRebuilt, 2021-01-12:
Maybe i've missed something in the configuration but i cannot run properly the modules with the nanoleaf mode.
The color waves and solid color works without a problem tho.

Can someone help me?
This is my config section:
`#define LED_TYPE            WS2812B                     // You might also use a WS2811 or any other strip that is Fastled compatible 
#define DATA_PIN            5//D1                          // Be aware: the pin mapping might be different on boards like the NodeMCU
//#define CLK_PIN             D5                        // Only required when using 4-pin SPI-based LEDs
#define CORRECTION          UncorrectedColor            // If colors are weird use TypicalLEDStrip
#define COLOR_ORDER         GRB                         // Change this if colors are swapped (in my case, red was swapped with green)
#define MILLI_AMPS          16000                       // IMPORTANT: set the max milli-Amps of your power supply (4A = 4000mA)
#define VOLTS               5                           // Voltage of the Power Supply

//#define REMOVE_VISUALIZATION          // remove the comment to completly disable all udp-based visualization patterns

#define HOSTNAME "Hex-Leaf"                 // Name that appears in your network, don't use whitespaces, use "-" instead

#define DEVICE_TYPE 4                   // The following types are available`

`#if DEVICE_TYPE == 0                // Generic LED-Strip
    #define NUM_LEDS 21*21
    //#define NUM_LEDS 33
    //#define NUM_LEDS 183
    #define BAND_GROUPING    21            // Groups part of the band to save performance and network traffic`

`#elif DEVICE_TYPE == 4              // Nanoleafs
    #define LEAFCOUNT 21                // Amount of triangles
    #define PIXELS_PER_LEAF 21          // Amount of LEDs inside 1x Tringle`

ALSO (let me know if i have to create another thread for this) can this system work with nodeRed?

WarDrake, 2021-01-13:
So, if I understand, your problem is that youcan only use those two patterns?
what happens when you use any other pattern?
connect the esp to a computer and log any errors the system is throwing and paste those here

HexRebuilt, 2021-01-14:
yes exactly, everything works except the nanoleaf mode.

from the serial port i read this:
[image]
[image]
[image]

and if i change something on the webpage nothing happens in the serial comunication

bb-Ricardo, 2021-01-15:
Hi, can you check out the latest "in-development" branch? There we added more verbose logging and most of the UI actions should be logged.

jasonmacdonald, 2021-01-16:
I'm seeing the same issue. Everything works EXCEPT the Nanoleaf patterns. No LEDs come on at all for those, despite having it set to 13 leaves with 25 lights each.

HexRebuilt, 2021-01-18:
> 
> 
> Hi, can you check out the latest "in-development" branch? There we added more verbose logging and most of the UI actions should be logged.

`{$d▒▒|▒d▒<
▒
l▒c|▒▒▒{▒c▒
c▒▒no▒lgg▒▒▒cp▒▒l{lslp▒g▒▒
l
▒▒c
g▒|▒d▒
▒c▒▒no▒l▒▒l`▒g
d`gr▒▒▒ncl`s▒▒o
cd`▒#
▒▒${▒`▒o▒▒▒

           DEBUG [191]: Loading config
                                      Heap: 34360
Boot Vers: 6
CPU: 80
SDK: 2.2.2-dev(38a443e)
Chip ID: 15535833
Flash ID: 1458280
Flash Size: 4194304
Vcc: 65535

*WM: [1] getCoreVersion():          2_7_4
*WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
*WM: [1] system_get_boot_version(): 6
*WM: [1] getFreeHeap():             31864
*WM: [1] AutoConnect
*WM: [3] setupHostname:  LEDs
*WM: [2] Setting WiFi hostname
*WM: [2] Connecting as wifi client...
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping
*WM: [1] Connecting to SAVED AP: MyWiFi
*WM: [3] Using Password: XXXXXXXXXXXXXXXX
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS
*WM: [1] STA IP Address: 192.168.0.33
*WM: [3] hostname: STA:  LEDs
INFO: Wi-Fi connected
DEBUG [1001]: SPIFFS contents:
                              DEBUG [1010]: FS File: /css/bootstrap.min.css, siz                                                                             e: 121099
         DEBUG [1010]: FS File: /css/dark-mode.css, size: 1166
                                                              DEBUG [1018]: FS F                                                                             ile: /css/minicolors.min.css, size: 96597
                                         DEBUG [1019]: FS File: /css/styles.css,                                                                              size: 2243
           DEBUG [1023]: FS File: /favicon.ico, size: 4286
                                                          DEBUG [1030]: FS File:                                                                              /fonts/glyphicons.woff2, size: 18028
                                     DEBUG [1033]: FS File: /images/atom196.png,                                                                              size: 3889
           DEBUG [1038]: FS File: /images/github.ico, size: 6518
                                                                DEBUG [1042]: FS                                                                              File: /images/surr.png, size: 1897
                                   DEBUG [1047]: FS File: /index.htm, size: 9822
                                                                               D                                                                             EBUG [1051]: FS File: /js/app.js, size: 16598
                                             DEBUG [1056]: FS File: /js/bootstra                                                                             p.min.js, size: 37051
                     DEBUG [1059]: FS File: /js/dark-mode-switch.min.js, size: 4                                                                             93
  DEBUG [1069]: FS File: /js/jquery-3.1.1.min.js, size: 86713
                                                             DEBUG [1071]: FS Fi                                                                             le: /js/minicolors.min.js, size: 15473
                                      DEBUG [1075]: FS File: /js/r-websocket.min                                                                             .js, size: 3101
               DEBUG [1080]: FS File: /settings.htm, size: 10090
                                                                DEBUG [1085]: FS                                                                              File: /simple.htm, size: 3745

                              DEBUG [1112]: FS Size: 1953282KB, used: 447784KB,                                                                              22.92%

      DEBUG [1112]: Enabled Features:
[... 22 lines cut ...]
                                                                                                                                                            DEBUG [70081]: Setting: pattern: Nanoleaf Rainbow Band Visualizer
                                                                DEBUG [71005]: Saving Config
                                                                                            DEBUG [71239]: Heap: 27440
                                                                                                                      DEBUG [73302]: Setting: pattern: Nanoleaf Solid Color Band Visualizer
                              DEBUG [75364]: Setting: pattern: Solid Color
                                                                          DEBUG [79001]: Setting: solid Color: red 255, green 255, blue 0
                                                                                                                                         DEBUG [81020]: Saving Config
        DEBUG [81253]: Heap: 27344
                                  DEBUG [86479]: Setting: pattern: Pride
                                                                        DEBUG [88937]: Setting: pattern: Color Waves
                                                                                                                    DEBUG [91000]: Saving Config
[... cut ...]
                                                                                                                                           DEBUG [131018]: Heap: 27440
         DEBUG [141009]: Heap: 27440
                                    DEBUG [151022]: Heap: 27440
                                                               DEBUG [151284]: Setting: pattern: Color Waves
                                                                                                            DEBUG [161011]: Heap: 27440

`

that's all i can see in the serial monitor. as before when i set the nanoleaf mode every led is turned off.

bb-Ricardo, 2021-01-18:
The Nanoleaf pattern only work if "ENABLE_UDP_VISUALIZATION" is used. These work only with the visualisation software mentioned in the README. I fixed it and they should not be visible in the current "in-development" branch anymore.

Sorry, this was a mistake. The Nanoleaf pattern should have not been visible.

HexRebuilt, 2021-01-26:
okay so the nanoleaf pattern is not ready yet. In that case i will run them in color waves even tho would be nice to change the speed of it.
I think that this issue can be closed. thanks a lot!

closed 2021-01-26 by HexRebuilt