NimmLor/esp8266-fastled-iot-webserver #122: Commands sent from web are not processed through

coolkv, 2021-01-13:
Flashed and setup everything correctly as below;
1. Using D1 Mini V2 NodeMcu and WS2812B 60Led/m.
2. Installed all latest drivers for ESP8266, FastLED and WiFiManager.
3. Correct LED_Type and datapins have been defined along with proper power data.
3. Both sketch and ESP sketch have been uploaded.
3. Able to connect with the esp via 192.168.4.1, Configured to use the wifi AP and save. 
4. Able to update the wifi connections & also reboot from the settings menu. 
5. Powered with 5V 3A external power adaptor 

**Issues**: 
Unable to toggle power/brightness. 
Unable to control the LEDs (RGB, color palettes, patterns, solid color, none) 
None of the options on the home menu work (Autoplay, solid color, color patterns etc)

Approaches to resolve:
Tried powering only one LED via USB and also powered externally. 
changed the Data pin on the esp, also other LED profiles and color order (WS2811/RGB/GRB/BGR)
LEDs are powered on and work. Light on the D1 is also on. 
The moment data-in/GND is connected, all connected LEDs power on in RED and no other change happens.

WarDrake, 2021-01-14:
try putting a 1k resistor in the data line between the first led and the Arduino... your board might not be cleaning the signal or your internal resistor might be damaged...

bb-Ricardo, 2021-01-15:
Can you try out the latest "in-development" branch? There you can see if your D1 even receives the commands. Keep your Serial Monitor open

coolkv, 2021-01-15:
> Can you try out the latest "in-development" branch? There you can see if your D1 even receives the commands. Keep your Serial Monitor open

Will try and post here.

coolkv, 2021-01-15:
> Can you try out the latest "in-development" branch? There you can see if your D1 even receives the commands. Keep your Serial Monitor open

This is so much better. I can see the commands are sent from mobile and received by the D1. guess the LEDs are broken. Ordered new set. will post updates.. 
Will reopen issue if it doesnt work. Thanks for the reply and the fantastic codes..!

coolkv, 2021-01-16:
Greetings! 
the ws2812bs now respond to changes made from the web applet..!
although while uploading the sketch and flashing the esp, i noticed that there is an open command that says "Hard resetting via RTS pin..." but i figured this might just be a warning and went ahead. (see image below)
Arduino says that it is done uploading the sketch and I reset the board once. and then flashed the ESP sketch. SPIFFS flashed successfully.

The serial monitor displays the commands sent over the applet,

And yes, i've added a 1k 0.25W resistor (Any 1k resistor would do it seems) to the data in. 
Found out my board was a clone and D4 was used for flashing. Data pin changed to D6.

Did all the above and now it works! thanks for all the comments, It guided me towards this! 👍

closed 2021-01-16 by coolkv

bb-Ricardo, 2021-01-16:
Great to hear.

> Found out my board was a clone and D4 was used for flashing. Data pin changed to D6.

Ohh, interesting. Something we have to add to the questions if a board is not working.

> Did all the above and now it works! thanks for all the comments, It guided me towards this! 👍

You're welcome 😄