NimmLor/esp8266-fastled-iot-webserver #123: Can't compile for WS2801 on Wemos

PatrickSteiner, 2021-01-14:
Hi,

I'm sure it's with me, but I can't seem to find the right config to build for WS2801 on Wemos D1mini

Even when using

#define FASTLED_ESP8266_D1_PIN_ORDER
#include <FastLED.h>

...

#define LED_TYPE            WS2801                     // You might also use a WS2811 or any other strip that is Fastled compatible 
#define DATA_PIN            D3      

I still get

../libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified 
static_assert(validpin(), "Invalid pin specified");

Any help is appreciated

bb-Ricardo, 2021-01-14:
Hi D3 doesn't exist on WEMOS D1. Try PIN `2`. This is D4 on the PCB, directly next to Ground.

PatrickSteiner, 2021-01-14:
Still getting the same error ... adding the entire stacktrace

/Users/psteiner/Documents/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002
 #    pragma message "FastLED version 3.003.002"
                     ^
In file included from /Users/psteiner/Documents/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /Users/psteiner/Documents/Arduino/Projects/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino:23:
/Users/psteiner/Documents/Arduino/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /Users/psteiner/Documents/Arduino/libraries/FastLED/FastLED.h:48:0,
                 from /Users/psteiner/Documents/Arduino/Projects/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino:23:
/Users/psteiner/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<66u>':
/Users/psteiner/Documents/Arduino/libraries/FastLED/fastspi_bitbang.h:23:50:   required from 'class AVRSoftwareSPIOutput<2u, 66u, 6u>'
/Users/psteiner/Documents/Arduino/libraries/FastLED/fastspi.h:34:7:   required from 'class SPIOutput<2u, 66u, 6u>'
/Users/psteiner/Documents/Arduino/libraries/FastLED/chipsets.h:157:6:   required from 'class LPD6803Controller<2u, 66u, (EOrder)10u, 6u>'
/Users/psteiner/Documents/Arduino/libraries/FastLED/FastLED.h:239:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)2u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u]'
/Users/psteiner/Documents/Arduino/Projects/esp8266-fastled-iot-webserver/esp8266-fastled-iot-webserver.ino:709:68:   required from here
/Users/psteiner/Documents/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

bb-Ricardo, 2021-01-14:
Did you set `D2` or just `2`?

PatrickSteiner, 2021-01-14:
yes!

```
/*######################## MAIN CONFIG ########################*/
#define LED_TYPE            WS2801                     // You might also use a WS2811 or any other strip that is Fastled compatible 
#define DATA_PIN            2                          // Be aware: the pin mapping might be different on boards like the NodeMCU
#define CLK_PIN             3                         // Only required when using 4-pin SPI-based LEDs
```
also not sure on CLK_PIN if that is needed with WS2801, as they do have 4 wires ( am more a Software than HW guy )

bb-Ricardo, 2021-01-14:
I'm not sure if a WEMOS D1 is compatible with a WS2801. SPI (not sure what this is) is needed for WS2801 and the D1 doesn't seem to have it.

NimmLor, 2021-01-19:
Hi, for SPI based strips you have to use the following configuration.
```
#define DATA_PIN           D1
#define CLK_PIN             D2
```
The Wemos D1 mini only supports SPI on D2/D1 (aka GPIO4/GPIO5). Image for reference:

[image]

closed 2021-01-21 by PatrickSteiner