NimmLor/esp8266-fastled-iot-webserver #126: payload syntax

The-Holgi, 2021-01-14:
Hello,
I‘ m using fhem for home automation.
I would serve the led‘s with MQTT.
The state of the led‘s where display in fhem, but how can i serve the led‘s?

WarDrake, 2021-01-14:
For MQTT the topic is : homeassistant/light/nanoleafs  
the last part being conditional on your settings in the firmware  

to change the values using a message the message is sent to : topic/set
ex: homeassistant/light/nanoleafs/set

a payload example to load the pride effect is:  
{"state": "ON", "effect": "Pride"}  

a payload to set the nanoleafs as a red color is:
{"state": "ON", "color": {"r": 255, "g": 72, "b": 118}}

to change a pattern the only required ones are `state` and `effect` but you can send as many commands together as you like

This one for example would set the state to on, brightness to 31% (80/255), pattern to Horizontal Rainbow, Autoplay to off, and Speed to 63
{"state":"ON","brightness":80,"effect":"Horizontal Rainbow","autoplay":0,"speed":63}

bb-Ricardo, 2021-01-14:
Is this possible to document in the README?

WarDrake, 2021-01-14:
Definitelly, I'll add it later today

PR #137 (2021-01-20): MQTT payload documentation
Adding MQTT commands documentation to the readme file
closes #126
closed 2021-01-20 by WarDrake