NimmLor/esp8266-fastled-iot-webserver #127: Reboot power on

arnaud-42, 2021-01-15:
Hello,

I was wondering why some mornings I find my leds powered on. When a reboot happen (i don't know why it reboot yet) the leds are powered on.
Is there any way to change this default behaviour ?

bb-Ricardo, 2021-01-15:
Hi,

you like to have the LEDs off by default at reboot?

The last power state is save and resumes after reboot.

arnaud-42, 2021-01-15:
are you sure ?
I just tried with the led off, i unplugged it and plug it back and the light turned on.

bb-Ricardo, 2021-01-15:
Just try out latest "in-development". Keep your serial monitor open.

It takes 10 seconds after the latest config change to save the date. Once the data is saved you sould reset the device and it should stay off.

If not, then it's a bug and I'll fix it asap.

arnaud-42, 2021-01-15:
Thanks i'll try tommorow and report here.

arnaud-42, 2021-01-16:
Just tried with the in-dev and it's the same, when i unplugged and plugged in back it turned ON.

Just to let you know, my esp are powered via usb.

bb-Ricardo, 2021-01-16:
Yes, you are correct. This is a bug. Just found the problem. Will fix it later. Thanks for reporting

bb-Ricardo, 2021-01-17:
Hi

can you test the "fix-power-off-mode" branch and see if this fixes it?

Thank you

arnaud-42, 2021-01-17:
It's working.
Thanks.

bb-Ricardo, 2021-01-17:
Thank you for testing. Merged to "in-development" branch

commit 829102be (2021-01-17): Merge branch 'fix-power-off-mode' into in-development

fixes: #127
closed 2021-01-17 by bb-Ricardo