NimmLor/esp8266-fastled-iot-webserver #131: Android/(iOS) App

NimmLor, 2021-01-17:
Hi, I'm thinking of building an react-native app to control the LEDs.

React-Native is a framework that allows to build hybrid apps that run on android, iOS and even on the web. However, I'm not planning to release to the app store because of the yearly fees (~ 125$/y).

In the first stage I'm planning to implement just the basic control UI.

Your welcome to post some feedback and suggestions here 😄

bb-Ricardo, 2021-01-17:
Hey,

you could do that but I think this is a bit too much. It would definitely integrate much nicer but also takes quite some time to develop. And I would need to compile it myself to get it onto my phone.

I was searching for generic LED control app where we would just need to implement the endpoint but couldn't find anything.

Also Blynk sounds good but this requires to use an IoT cloud or running your own instance locally.

I think the current solution is quite ok. Just needs some UI improvements. And Uploading the data to the filesystem should also be improved, that you don't need the Arduino plugin (just don't know how).

Maybe loading from github:
`if (wifiConnected && SPIFFS.bytesUsed == 0) { getDataFromGithub(); }` or something like that.

bb-Ricardo, 2021-01-17:
There is another popular (W)LED webserver project on github which has a good color picker and pattern controls. But the settings pages are really ugly. Maybe this has some inspirations on improving things.

WarDrake, 2021-01-18:
I've been using homeassistant to contol mine, I never even open the web settings anymore, homeassistant is built on react and it's a great way at least for me control it I have the UI I like.

also, the app you wouldn't need to compile Ricardo, it could simple be added as a different git repo with it's own releases and you just download an APK that's open source.

as for endpoints, anything that can send MQTT messages can control the leds using the mqtt integration already, tho webhooks could be easily added.

CrimsonAC, 2021-01-18:
In my case, i am planning to control the LEDs via MQTT, too. 
So in my opinion there is no need for a dedicated app to control the LEDs.
The webserver should be sufficient for someone who does not use automation.

NigaMan, 2021-01-20:
and as for me, it would be cool to create a separate application, so as not to go into the browser, but just through the phone turned on and everything, just through the phone you can connect several panels.
it would also be cool to make an implementation of this type, so that through the phone you can create a diagram of your panels.
[image]

philippnbg, 2021-07-09:
@WarDrake how did you integrate the endpoint to homeassistant and how do you control the various settings via the homeassistant-frontend? so far i did not find any custom integrations (e.g. via HACS), nor natively