NimmLor/esp8266-fastled-iot-webserver #133: Only first leaf lights up

passteinforth, 2021-01-18:
Hi there,

my Problem is, that only my first leaf shines up, and the second one is dark. Right now I changed the following in the Code:
 #define DATA_PIN            D3          // my actual data pin
#define MILLI_AMPS          10000
#define DEVICE_TYPE 4 

#elif DEVICE_TYPE == 4              // Nanoleafs
    #define LEAFCOUNT 2                // Amount of triangles
    #define PIXELS_PER_LEAF 12          // Amount of LEDs inside 1x Triangle

I checked my solderred connection, at the end of the strip there are 5.4 V present. D= from the last strip of leaf 1 has a verified solder connection to the first strip of leaf two. I once connected the data pin from the wemos directly to leaf 2 in parallel and the LEDs went on (static though, because I just bridged them with a cable without soldering). 
I also followed the arrows, which are pointing away from my board. 
The web UI works perfectlly fine for the first leaf, I can change everything and it works, just my second leaf won't show up.

Strangely, I changed to DEVICE_TYPE 0 for a try and gave it a number of 16 LEDs (had only 4 *4 LEDs connected then), but only the first 3*4 LEDs turned on. All of them should glow even without the web UI, right?

I added some code to print out DEVICE_TYPE and NUM_LEDs in the serial monitor, and by this I can say, the flash was successful. 

Do you have a clue for this? Alll solder connections seem to be fine, but also the code works for the first leaf. It is like it remember my first ever test config, where I gave him only LEAFCOUNT 1. Do I have to erase somehow the old code before? Or should I flash something else in between, so he forgets the old code?

bb-Ricardo, 2021-01-18:
Hi.

I tested it and it worked for my setup.

I assume you have a WEMOS 1D? I have the same board but the PIN description on the board does not match the PINS in the ARDUINO IDE. I have to set DATA_PIN to `2` if I want to use PIN `D4`. Can you try the same and see if this works?

passteinforth, 2021-01-19:
Thanks for the fast answer and sorry, I completely forgot to write down my Setup. I use a WEMOS D1 Mini and a 10 A 5V power supply (which worked in an ambilight project before with over 150 LEDs). Additionally, I also use the recommended logic level shifter to boost the data signal from 3.3 V to 5 V.

I think my previous configuration of DATA_PIN (D3) should be right, because I could change color, brightness, etc in the web UI. However, I tried soldering the Data Pin to D4 on the board and set it to 2 in the sketch. This lead to unresponsive behaviour of only the first strip in the first LED. When setting DATA_PIN to D4, I could again change color etc in the UI, but still only the first leaf lights up. 
I will look for a not cutted piece of LEDs to be sure its not about my soldering. It's rather strange as I can check the soldering which seems fine, but also the code works for some LEDs and the Code writes the right NUM_LEDS in the serial monitor.

WarDrake, 2021-01-19:
While you're physically checking, make sure you didn't put one of the pieces of cut leds backwards and the data line is inverted, that would explain your issues.

passteinforth, 2021-01-19:
Just minutes ago, I found my error. And it's a little bit embarassing one though... Well, however, the code is perfectly fine, so I will close this issue. 
Also my solder connections were fine, but I damaged the little resistor (?) between the LED and the Data Out pin of my LED strip. therefore the Data signal was not forwarded through the wired connection.
Thanks both of you!

closed 2021-01-19 by passteinforth