NimmLor/esp8266-fastled-iot-webserver #134: Compile Erroe with Board NodeMcu

Jumper2055, 2021-01-18:
Hi guys

i can´t compile the latest master code for my NodeMcu

Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

In file included from \Documents\Arduino\LEDSERVEREDIT\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:1:0:

\Documents\Arduino\libraries\WiFiManager-development/strings_en.h:375:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};

       ^

\Documents\Arduino\libraries\WiFiManager-development/strings_en.h:376:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,WIFI_COUNTRY_POLICY_AUTO};

       ^

\Documents\Arduino\libraries\WiFiManager-development/strings_en.h:377:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,WIFI_COUNTRY_POLICY_AUTO};

       ^

In file included from \Documents\Arduino\LEDSERVEREDIT\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:23:0:

\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003

 #    pragma message "FastLED version 3.003.003"

                     ^

In file included from \Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 \Documents\Arduino\LEDSERVEREDIT\esp8266-fastled-iot-webserver\esp8266-fastled-iot-webserver.ino:23:

\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

exit status 1

Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Can anyone help

bb-Ricardo, 2021-01-18:
Hi,

It seems to be an error in the wifiManager library. Can you try to checkout version 2.0.3 and see if the issue persists?

Jumper2055, 2021-01-19:
Hey Ricardo,
thank you for your answer. I set the version of the library to  2.0.3 an now i can  compile the build.

after that i have an other problem. The webui say only Connecting, please wait. The Wlan is connected.
[image]

I use a AZ-Delivery D1Mini NodeMcu with the D4 pin (in Arduino 2). The build from december 2020 has woked fine.

I have no idea

Thank you

bb-Ricardo, 2021-01-19:
Hey,

This looks like an older version of the web interface. Try to reupload the data dir (check the docs for instructions).

Jumper2055, 2021-01-19:
Hey. 
It was a chache error of my browser.

But another problem.

When i connect the node to my stripe it only lights up the first led in yellow.

I dont know

Any ideas?

Jumper2055, 2021-01-19:
Oh my god.

I have used the wrong board.

Its finaly working now.

Thank you

closed 2021-01-19 by Jumper2055