NimmLor/esp8266-fastled-iot-webserver #135: Duplicate open SSID after Wifimanager connected

jakesza, 2021-01-19:
Hi,

I have a few nanoleafs in the house and updated with the latest branch but have a few issues;

biggest issue: once Wemos D1 Mini flashed, 2 SSID's pops up, nanoleaf-E470 and ESP-E470, So once Wifimanager completed and reboots the deivce is accessible from the lan but leaves an open SSID that I cannot get rid off. I have reflashed with different settings, mDNS. still exists. please help.
It also seems that if you leave the MQTT settings as is pre flash, you cannot change it afterwards, yes sure, you can change the settings on settings page, but the real fact is they stay as they were pre flashed.

Output:

19:36:19.417 -> Heap: 37200
19:36:19.417 -> Boot Vers: 31
19:36:19.417 -> CPU: 80
19:36:19.417 -> SDK: 2.2.2-dev(38a443e)
19:36:19.417 -> Chip ID: 5126513
19:36:19.417 -> Flash ID: 1458415
19:36:19.417 -> Flash Size: 4194304
19:36:19.417 -> Vcc: 65535
19:36:19.463 -> 
19:36:19.463 -> SPIFFS contents:
19:36:19.509 -> FS File: /css/bootstrap.min.css, size: 121093
19:36:19.509 -> FS File: /css/dark-mode.css, size: 1104
19:36:19.509 -> FS File: /css/minicolors.min.css, size: 96597
19:36:19.509 -> FS File: /css/styles.css, size: 2152
19:36:19.509 -> FS File: /favicon.ico, size: 4286
19:36:19.509 -> FS File: /fonts/glyphicons.woff2, size: 18028
19:36:19.509 -> FS File: /images/atom196.png, size: 13240
19:36:19.509 -> FS File: /images/github.ico, size: 6518
19:36:19.509 -> FS File: /images/surr.png, size: 1897
19:36:19.509 -> FS File: /index.htm, size: 9600
19:36:19.509 -> FS File: /js/app.js, size: 16035
19:36:19.509 -> FS File: /js/bootstrap.min.js, size: 37045
19:36:19.509 -> FS File: /js/dark-mode-switch.min.js, size: 492
19:36:19.555 -> FS File: /js/jquery-3.1.1.min.js, size: 86709
19:36:19.555 -> FS File: /js/minicolors.min.js, size: 15463
19:36:19.555 -> FS File: /js/r-websocket.min.js, size: 3100
19:36:19.555 -> FS File: /settings.htm, size: 9862
19:36:19.555 -> FS File: /simple.htm, size: 3717
19:36:19.555 -> 
19:36:19.555 -> *WM: [1] AutoConnect 
19:36:19.555 -> *WM: [3] setupHostname:  Nanoleaf-3
19:36:19.555 -> *WM: [2] Setting WiFi hostname 
19:36:19.555 -> *WM: [2] Connecting as wifi client... 
19:36:19.555 -> *WM: [3] STA static IP:
19:36:19.555 -> *WM: [2] setSTAConfig static ip not set, skipping 
19:36:19.602 -> *WM: [1] Connecting to SAVED AP: AC88U
19:36:19.602 -> *WM: [3] Using Password:
19:36:19.602 -> *WM: [3] WiFi station enable 
19:36:19.602 -> *WM: [3] enableSTA PERSISTENT ON 
19:36:19.602 -> scandone
19:36:19.602 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
19:36:22.424 -> scandone
19:36:22.424 -> no AC88U found, reconnect after 1s
19:36:22.517 -> reconnect
19:36:25.390 -> scandone
19:36:25.390 -> no LNSK_AC88U found, reconnect after 1s
19:36:25.482 -> reconnect
19:36:28.310 -> scandone
19:36:28.310 -> state: 0 -> 2 (b0)
19:36:28.356 -> state: 2 -> 3 (0)
19:36:28.448 -> state: 3 -> 5 (10)
19:36:28.448 -> add 0
19:36:28.448 -> aid 41
19:36:28.448 -> cnt 
19:36:29.557 -> 
19:36:29.557 -> connected with AC88U, channel 7
19:36:29.603 -> dhcp client start...
19:36:56.526 -> bcn_timout,ap_probe_send_start
19:36:57.310 -> ip:192.168.0.29,mask:255.255.255.0,gw:192.168.0.1
19:36:57.310 -> *WM: [2] Connection result: WL_CONNECTED
19:36:57.310 -> *WM: [3] lastconxresult: WL_CONNECTED
19:36:57.310 -> *WM: [1] AutoConnect: SUCCESS 
19:36:57.310 -> *WM: [1] STA IP Address: 192.168.0.29
19:36:57.310 -> *WM: [3] hostname: STA:  Nanoleaf-3
19:36:57.310 -> Wi-Fi connected
19:36:57.356 -> HTTP web server started
19:36:58.238 -> Connected! Open http://192.168.0.29 in your browser
19:37:18.877 -> bcn_timout,ap_probe_send_start
19:37:27.896 -> bcn_timout,ap_probe_send_start
19:37:36.889 -> bcn_timout,ap_probe_send_start

bb-Ricardo, 2021-01-19:
Hi,

can you please check out the lastest "in-development" branch? Quite a few things got changed/fixed.

jakesza, 2021-01-19:
Thank you, just reflashed with the in dev branch, same results, 2 SSID's, the main mDNS SSID goes away once connected, but the ESP SSID is still active and you can connect to the internal ip and use the web server controls.

bb-Ricardo, 2021-01-19:
So the esp is connected to the wireless and providing an acess point at thw same time?

Can you paste the output of the serial monitor here?

jakesza, 2021-01-19:
> So the esp is connected to the wireless and providing an acess point at thw same time?
> 
> Can you paste the output of the serial monitor here?

correct. See below;

22:36:04.833 -> 
22:36:04.833 -> DEBUG [173]: Loading config
22:36:04.833 -> 
22:36:04.833 -> DEBUG [175]: System Information:
22:36:04.833 -> DEBUG [175]: Heap: 36808
22:36:04.833 -> DEBUG [175]: Boot Vers: 31
22:36:04.833 -> DEBUG [175]: CPU Speed: 80 MHz
22:36:04.833 -> DEBUG [175]: SDK: 2.2.2-dev(38a443e)
22:36:04.833 -> DEBUG [178]: Chip ID: 5126513
22:36:04.833 -> DEBUG [181]: Flash ID: 1458415
22:36:04.833 -> DEBUG [184]: Flash Size: 4194304KB
22:36:04.833 -> DEBUG [187]: Vcc: 65535
22:36:04.833 -> 
22:36:04.879 -> *WM: [1] getCoreVersion():          2_7_4
22:36:04.926 -> *WM: [1] system_get_sdk_version():  2.2.2-dev(38a443e)
22:36:04.926 -> *WM: [1] system_get_boot_version(): 31
22:36:04.926 -> *WM: [1] getFreeHeap():             34272
22:36:04.926 -> *WM: [1] AutoConnect 
22:36:04.926 -> *WM: [3] setupHostname:  Nanoleaf-1
22:36:04.926 -> *WM: [2] Setting WiFi hostname 
22:36:04.926 -> *WM: [2] Connecting as wifi client... 
22:36:04.926 -> *WM: [3] STA static IP:
22:36:04.926 -> *WM: [2] setSTAConfig static ip not set, skipping 
22:36:04.926 -> *WM: [1] Connecting to SAVED AP: AC88U
22:36:04.926 -> *WM: [3] Using Password
22:36:04.926 -> *WM: [3] WiFi station enable 
22:36:04.926 -> *WM: [3] enableSTA PERSISTENT ON 
22:36:04.926 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
22:36:05.019 -> *WM: [2] Connection result: WL_CONNECTED
22:36:05.019 -> *WM: [3] lastconxresult: WL_CONNECTED
22:36:05.019 -> *WM: [1] AutoConnect: SUCCESS 
22:36:05.019 -> *WM: [1] STA IP Address: 192.168.0.29
22:36:05.019 -> *WM: [3] hostname: STA:  Nanoleaf-1
22:36:05.019 -> INFO: Wi-Fi connected
22:36:05.019 -> DEBUG [355]: SPIFFS contents:
22:36:05.019 -> DEBUG [363]: FS File: /css/bootstrap.min.css, size: 121093
22:36:05.019 -> DEBUG [364]: FS File: /css/dark-mode.css, size: 1350
22:36:05.019 -> DEBUG [371]: FS File: /css/minicolors.min.css, size: 96597
22:36:05.019 -> DEBUG [372]: FS File: /css/styles.css, size: 1392
22:36:05.065 -> DEBUG [377]: FS File: /favicon.ico, size: 4286
22:36:05.065 -> DEBUG [382]: FS File: /fonts/glyphicons.woff2, size: 18028
22:36:05.065 -> DEBUG [386]: FS File: /images/atom196.png, size: 13240
22:36:05.065 -> DEBUG [391]: FS File: /images/github.ico, size: 6518
22:36:05.065 -> DEBUG [396]: FS File: /index.htm, size: 9363
22:36:05.065 -> DEBUG [400]: FS File: /js/app.js, size: 16087
22:36:05.065 -> DEBUG [404]: FS File: /js/bootstrap.min.js, size: 37045
22:36:05.065 -> DEBUG [408]: FS File: /js/dark-mode-switch.min.js, size: 492
22:36:05.065 -> DEBUG [417]: FS File: /js/jquery-3.1.1.min.js, size: 86709
22:36:05.065 -> DEBUG [419]: FS File: /js/minicolors.min.js, size: 15463
22:36:05.112 -> DEBUG [423]: FS File: /settings.htm, size: 9456
22:36:05.112 -> 
22:36:05.112 -> DEBUG [450]: FS Size: 1953282KB, used: 445023KB, 22.78%
22:36:05.112 -> 
22:36:05.112 -> DEBUG [451]: Enabled Features:
22:36:05.112 -> DEBUG [451]: Feature: mDNS support enabled
22:36:05.112 -> DEBUG [451]: Feature: OTA support enabled
22:36:05.112 -> 
22:36:05.158 -> INFO: HTTP web server started
22:36:05.668 -> INFO: WiFi Connected! Open http://192.168.0.29 in your browser
22:36:05.668 -> INFO: mDNS responder started. Try to open http://Nanoleaf-1.local in your browser
22:36:05.668 -> DEBUG [1008]: Stats: 45 frames/s, frame delay: 4250 us

bb-Ricardo, 2021-01-19:
This seems very odd. And that there are more or other devices which are maybe creating these APs?

This ESP is running in Client mode.

You could use a computer and check the mac addresses of these APs.

And if you unplug your ESP all these APs disappear?

Did you try deleting the wifi settings?

jakesza, 2021-01-20:
> This seems very odd. And that there are more or other devices which are maybe creating these APs?
> 
> This ESP is running in Client mode.
> 
> You could use a computer and check the mac addresses of these APs.
> 
> And if you unplug your ESP all these APs disappear?
> 
> Did you try deleting the wifi settings?

I have all my ESP's setup with DHCP reservations in my network, so I knw the mac addresses, if you unplug the ESP, the AP disappears. Like I said, when you conect to the ESP AP, and open 192.168.4.1 it actually redirect to the Nanoleaf ui and not wifi manager. I flashed 3 different Wemos boards all with the same results. 

1. flash settings: erase all, board type Lolin Wemos D1 R1 mini
2. boot pins D3 and GND shorted
3. flash and reboot
4. sketch data upload and reboot
5. 2 AP's appear, Nanoleaf-1-xxxx and ESP-xxxx

Below only changes made in sketch:

#define DATA_PIN            D4
#define DEFAULT_HOSTNAME "Nanoleaf-1"
#define LED_DEVICE_TYPE 4
#elif LED_DEVICE_TYPE == 4              // Nanoleafs
    #define LEAFCOUNT 12                // Amount of triangles
    #define PIXELS_PER_LEAF 6          // Amount of LEDs inside 1x Tringle

#define ENABLE_OTA_SUPPORT                // requires ArduinoOTA - library, not working on esp's with 1MB memory (esp-01, Wemos D1 lite ...)
        #define OTA_PASSWORD "passwd123"      //  password that is required to update the esp's firmware wireless

    #define ENABLE_MULTICAST_DNS              // allows to access the UI via "http://<HOSTNAME>.local/", implemented by GitHub/WarDrake

    #define RANDOM_AUTOPLAY_PATTERN             // if enabled the next pattern for autoplay is choosen at random
    #define AUTOPLAY_IGNORE_UDP_PATTERNS        // remove visualization patterns from autoplay

#elif LED_DEVICE_TYPE == 4
        #define MQTT_TOPIC "light/nanoleaf-2"              // MQTT Topic to Publish to for state and config (Home Assistant)
        #define MQTT_TOPIC_SET "/set"                                   // MQTT Topic to subscribe to for changes(Home Assistant)
        #define MQTT_DEVICE_NAME "nanoleaf-2"

jakesza, 2021-01-20:
Wifi settings reset, here you can actually see the 2 AP;s on the same MAC.

[image]

NimmLor, 2021-01-20:
Hi, have you update the esp8266 board library to the latest version?

Another thing I would recommend to try is to set the erase mode to `All flash contents`:

[image]

jakesza, 2021-01-20:
> Hi, have you update the esp8266 board library to the latest version?
> 
> Another thing I would recommend to try is to set the erase mode to `All flash contents`:
> 
> [image]

Hi, I have 2.7.4 for esp8266 which is latest. I do believe its got something to do with wifimanager.

bb-Ricardo, 2021-01-20:
Can you post a screenshot of your Board configuration in Arduino?

jakesza, 2021-01-20:
> Can you post a screenshot of your Board configuration in Arduino?

[image]

arnaud-42, 2021-01-20:
I have the same issue here.
I have three boards ... and after they are connected to my IoT WIFI, I've got 3 open SSID starting by ESP-XXXXXX

Tom051276, 2021-01-21:
Hello, I have the same behaviour. I Thtink the problem is that the AP always gets up, not only if the connection to the WiFi fails, like it should. Regards Thomas

jakesza, 2021-01-23:
any ideas?

Tom051276, 2021-01-23:
I have worked around this issue.
For me it is working now, with the entry of WiFi.Mode(WIFI_STA) ;
before the WiFiManger.autoConnect()

[image]

jakesza, 2021-01-23:
> I have worked around this issue.
> For me it is working now, with the entry of WiFi.mode(WIFI_STA) ;
> before the WiFiManager.autoConnect()
> 
> [image]

Great find! Problem solved. tested all good now

Tom051276, 2021-01-23:
> > I have worked around this issue.
> > For me it is working now, with the entry of WiFi.mode(WIFI_STA) ;
> > before the WiFiManager.autoConnect()
> > [image]
> 
> Great find! Problem solved. tested all good now

Great to hear that. Thank you for the feeback

bb-Ricardo, 2021-01-31:
Hi,

can you test the latest "in-development" branch and see if the issue is gone?

Thank you

arnaud-42, 2021-01-31:
Working for me

bb-Ricardo, 2021-02-02:
Then I will close this issue as resolved. Thank you for testing

closed 2021-02-02 by bb-Ricardo