NimmLor/esp8266-fastled-iot-webserver #139: [Noob needs help] Wemos D1 and WS2812E not responding

PatrickSteiner, 2021-01-21:
Sorry to place this here, as it's not an issue but most likely my problem.

I have an older [Wemos D1](https://cdn-images-1.medium.com/max/955/0*JTRYFCNgFmBo2pVf.jpg) and WS2812E ( order 2812B and received the E ones ).

When starting the flashed Wemos 2 LEDs light up and I can't control anything. Not even turn them "off". So most likely my config is messed-up. 

This is what I have configured:

  - Board in ArduinoIDE:  LOLIN (WEMOS) D1 R2 & mini
  - DATA_PIN tried all from D1 to D4
  - Powered currently only via USB Port, as there are only 22 LEDS connected

If there is an obvious flaw in my thinking and implementation, please let me know.

bb-Ricardo, 2021-01-21:
I have mine configured with '2' in the scetch wich is 'D4' on the board. Juat try if this works.

PatrickSteiner, 2021-01-22:
@bb-Ricardo tried that with 2 different Wemos modules, no change.

There must be some config error that I'm doing, or incompatibility with the LOLIN Board Manager and this old Wemos?

bb-Ricardo, 2021-01-22:
I actually don't know. I ordered a bunch of them on ali a while ago and all of them worked without a flaw.

[image]
[image]

bb-Ricardo, 2021-02-02:
any updates/success

PatrickSteiner, 2021-02-02:
unfortunately not :-(

sp46887, 2021-07-30:
Perhaps I can help.
I bought a Wemos D1 mini clone  for my Thingiverse desk lamp, it is a
D1 Mini NodeMcu ESP8266-12F WIfi Modul from az-delivery
When I choose D1 R1 in Arduino boadmanager nothing works, only the first ws2812b LED is on, no control over the LEDs, I tried all possible ports.
Than I switched to “D1 mini clone” in Boardmanager, new compile and upload, now the lamp is fully functional, I use D4 pin for LED control. 
First LED “glitches”, a level shifter also solved this problem.

The bottom line is, switch to the right board in boardmanager.
I installed the newest available esp8266 support package into Arduino IDE

https://arduino.esp8266.com/stable/package_esp8266com_index.json